digistump/DigistumpArduino #3: IDE does not work on OSX 10.10 with java8

tcurdt, 2015-01-06:
While java 8 is the recommended java version for OSX 10.10, the IDE requires a java 6 runtime

> To open “Arduino IDE - Digistump Release 1.5.8B” you need to install the legacy Java SE 6 runtime.

digistump, 2015-01-06:
The stable/most compatible version of Arduino for 1.5.8 requires Java 6 - from the Arduino website:

> Mac OS X: ZIP file for Java 6
> Mac OS X: ZIP file for Java 7. NOTE: the ZIP for Java 7 is experimental and it works only on OSX 10.7 or greater. If you experience problems running it please use the Java 6 version.

We use the Java 6 version for maximum compatibility - because we use the Arduino binaries unmodified this dependency is tied to the Arduino IDE.

tcurdt, 2015-01-06:
@digistump so you reckon I raise this at the official Arduino release instead?

Maximum compatibility means PITA for anyone running a current version of OSX.

digistump, 2015-01-06:
@tcurdt Yeah - we really have to follow their lead, once we have automated builds setup in the next month or two then we will happily build our package for both their Java 6 and 7 versions - but we don't have the resources to build the whole IDE (and fix any breaks) for 8.

digistump, 2015-01-06:
Is 7 any better supported in a fresh load of 10.10? I can't remember if I installed it myself or not.

closed 2015-01-06 by digistump

tcurdt, 2015-01-06:
@digistump frankly speaking - I am not sure. I only have java 8 on my 10.10 system. (so far - I guess. grrr)

https://github.com/arduino/Arduino/issues/2384

I added my comment to this issue - but it doesn't look like one should hold his/her breath for this. Only the JNI stuff should be the problem though. Shipping the libs for the appropriate java versions isn't that hard. Too bad they don't get their act together.

digistump, 2015-01-06:
@tcurdt  That's what I thought - grr is right.

Yep I wouldn't hold my breath - but you might be happy to know that we'll be offering an alternative to the Arduino IDE in the near future - both web based and desktop with a similar look to Sublime Text and all kinds of goodies like proper syntax error highlighting courtesy of clang

tcurdt, 2015-01-06:
@digistump that sounds interesting! Can't wait.

Would also love a command line way to compile, debug and update the little guys. I guess/hope there is a way already - but haven't looked into it yet.

tcurdt, 2015-02-11:
@digistump apparently it's fixed now https://github.com/arduino/Arduino/issues/2384

digistump, 2015-02-11:
> @digistump <https://github.com/digistump> apparently it's fixed now
> arduino/Arduino#2384 <https://github.com/arduino/Arduino/issues/2384>

We will release a 1.6 version, likely next week - until then you can
install 1.6 and move the Resources/hardware/digistump folder (from inside
the app bundle) into it and it "should" work