digistump/DigistumpArduino #4: OS X 1.5.8c verification error on launch

davelnewton, 2015-03-21:
> "Arduino IDE - Digistump Release 1.5.8C.app" is damaged and can't be opened. You should move it to the trash.

System: Mac Air OS X 1.9.5 8G

1.5.8B has same problem. Is this no longer the IDE to use with the Digisparks?

Let me know what other info is required.

davelnewton, 2015-03-21:
May be the stub issue; closing until verified.

closed 2015-03-21 by davelnewton

dgdosen, 2015-04-01:
I get the same issue...

digistump, 2015-04-02:
This generally means you need to go enable unverified applications (see the
getting started guide on the wiki) or the extractor you used didn't like
the zip file. Works on 10.9 and 10.10 for me.

lukemc, 2015-04-12:
Unzipping the DigistumpIDE1.5.8COSX and running the .app gives me:

`Arduino IDE - Digistump Release 1.5.8C.app" is damaged and can't be opened. You should move it to the trash`

Mid 2014, 13"MBPr OSX 10.10.1 (14B25)

davelnewton, 2015-04-12:
The "unverified applications" thing worked for me, FWIW.