digistump/DigistumpArduino #5: 64-bit version lacks micronucleus

jockster, 2015-04-07:
Downloaded the 64-bit release from below URL and tried using it to compile code; https://github.com/digistump/DigistumpArduino/releases/download/v1.5.8C/DigistumpIDE1.5.8C-linux64.tar.gz

The compilation failed and complaned about micronucleus missing despite the binary being present in the correct path (hardware/digistump/avr/tools/). I tried changing permissions, but to no result. I managed to sort it out by downloading micronucleus with git and recompiling (requiring the `libusb-dev` package on debian) it and replacing the micronucleus binary.

It all seems to work now but I reckon that if you fix this it should save other folks some time.

antonio-m, 2015-04-10:
yeah, the binary included is 32 bit, not 64... I saw the same issue in their forums but no aswer there

digistump, 2015-06-23:
This has been fixed in the upcoming release for Arduino 1.6.5+

Thanks!
Erik

closed 2015-06-23 by digistump