digistump/DigistumpArduino #12: contains deprecated recipe.ar.pattern

mpflaga, 2015-11-03:
A recent change in 1.6.6 hourly caused compile failures of the DigiSpark Platform. I reported it and it has been resolved. 

https://github.com/arduino/arduino-builder/issues/42#issuecomment-152974363

Where the recent compile has the following warning prints

    Warning: platform.txt from core 'Digistump AVR Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
    Warning: platform.txt from core 'Digistump AVR Boards' contains deprecated recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mmcu={build.mcu} -o "{build.path}/{build.project_name}.elf" {object_files} "{build.path}/{archive_file}" "-L{build.path}" -lm, automatically converted to recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mmcu={build.mcu} -o "{build.path}/{build.project_name}.elf" {object_files} "{archive_file_path}" "-L{build.path}" -lm. Consider upgrading this core.

KahnSoft, 2015-11-14:
Agree, i take the same Messages, and the Program dosen't upload, communication is ok.

Warning: non trusted contribution, skipping script execution (C:\Users\Schulz\AppData\Local\Arduino15\packages\digistump\tools\micronucleus\2.0a4\post_install.bat)
Warning: platform.txt from core 'Digistump AVR Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.

smartynov, 2015-12-20:
Is anyone able to fix this issue? Or at least explain how it could be fixed?

digistump, 2016-02-12:
fixed thanks to a pull request - will be fixed in todays release

closed 2016-02-12 by digistump