digistump/DigistumpArduino #13: Issue with IDE 1.6.6

efi, 2015-11-25:
In 1.6.6 the DigisparkMouse and DigisparkJoystick examples fail with

> _[...]_\hardware\tools\avr\avr\include\avr\sleep.h:634:6: error: #error "No SLEEP mode defined for this device."

A problem like this is discussed (and solved?) elsewhere: https://github.com/SpenceKonde/ATTinyCore/issues/31

So it's domething about
``` C
#if defined(__AVR_ATtiny84__)
```
but a simple `#define __AVR_ATtiny84__` in the sketch did not do the trick for me.

Can someone please look into that issue?

efi, 2015-11-30:
Well, well, well. With a bit more knowledge about the chip on my Digispark, this would have occured to me earlier:
``` C
#define __AVR_ATtiny85__
```
as the first line in the sketch is sufficient to make the compile error go away and make the device function as HID compatible mouse.

"ATtiny84" was of course wrong and while it allowed a SLEEP mode to be found, some other functions were not executed as intended.

So to solve this issue for everyone, the exampled for DigiMouse and DigiJoystick have to be alterd with the above line before the `#include`s.

hodavame, 2015-12-11:
Thank you!!!
I also had that problem with the joystick library.

digistump, 2016-02-12:
I've added this define to the variant pins file for the board so that it doesn't need to be defined anywhere else - releasing today

efi, 2016-02-12:
Great, thanks!

closed 2016-02-12 by efi