digistump/DigistumpArduino #15: 64-bit

probonopd, 2016-01-10:
`/tools/micronucleus-2.0a4-linux64.tar.gz` is supposed to be 64-bit but actually contains a 32-bit binary, as you can see when you run ` file micronucleus-2.0a4/launcher`:

```
launcher: ELF 32-bit LSB executable, Intel 80386, version 1 (SYSV), 
dynamically linked, interpreter /lib/ld-linux.so.2, for GNU/Linux 2.6.24, BuildID[sha1]=4de23b7931a126b710dca8463647dd931937030c, not stripped`
```
This results in:
```
java.io.IOException: Cannot run program 
"/root/.arduino15/packages/digistump/tools/micronucleus/2.0a4/launcher": 
error=2, No such file or directory
```

when trying to upload, hence entirely making the Linux 64-bit build unusable.

probonopd, 2016-01-10:
On Ubuntu, `apt-get install lib32stdc++6` is a workaround that makes it run. But this would not be necessary if the binary was compiled for 64-bit in the first place.

probonopd, 2016-01-24:
On Fedora, `sudo yum -y install libstdc++.i686` is a workaround. However this is silly and really should be fixed. Where is the source code for `launcher`?

digistump, 2016-02-13:
Fixed in todays coming release

closed 2016-02-13 by digistump

probonopd, 2016-02-13:
Can't seem to see any change in https://github.com/digistump/DigistumpArduino/tree/master/tools and am still missing the source code

digistump, 2016-02-13:
Source code: github.com/digistump/avr-dummy - launcher is just this renamed to launcher

Fixed here: https://github.com/digistump/DigistumpArduino/releases/download/1.6.5a/micronucleus-2.0a4-linux64.tar.gz - forgot the binary was also in the repo, will commit that as well now.