digistump/DigistumpArduino #16: DigisparkCDC: Error "BOGUS urb xfer, pipe 3 != type 1" on Linux

probonopd, 2016-01-10:
I keep getting (lots of) "BOGUS urb xfer, pipe 3 != type 1" in dmesg on Linux.
When running the `DigisparkCDC > Print` sketch, then the characters on the pseudo-serial console are often garbled, the output looks like

`
TEST!
TEST!
TT
!T!

TTEST!
TEST!
TEST!
TESTS
S
S!
TEST!
TT

S!TET
!TEST!
TEST!
T
STT
!SS
`
Possibly this is caused by something crashing the kernel module?

```
me@host:~$ uname -a

Linux host 4.2.0-16-generic #19-Ubuntu SMP Thu Oct 8 15:35:06 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

me@host:~$ cat /etc/*release

DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=15.10
DISTRIB_CODENAME=wily
DISTRIB_DESCRIPTION="Ubuntu 15.10"
NAME="Ubuntu"
VERSION="15.10 (Wily Werewolf)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 15.10"
VERSION_ID="15.10"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"

me@host:~$ dmesg

[  +0.001853] ------------[ cut here ]------------
[  +0.000005] WARNING: CPU: 0 PID: 0 at /build/linux-AxjFAn/linux-4.2.0/drivers/usb/core/urb.c:450 usb_submit_urb.part.6+0x142/0x550()
[  +0.000002] usb 10-1: BOGUS urb xfer, pipe 3 != type 1
[  +0.000001] Modules linked in: cdc_acm dell_wmi sparse_keymap gpio_ich video dcdbas snd_hda_codec_analog snd_hda_codec_generic coretemp kvm_intel snd_hda_codec_hdmi snd_hda_intel kvm snd_hda_codec snd_usb_audio serio_raw snd_hda_core snd_usbmidi_lib snd_hwdep cm109 snd_pcm snd_seq_midi snd_seq_midi_event snd_rawmidi input_leds snd_seq usblp snd_seq_device snd_timer lpc_ich snd mei_me mei 8250_fintek soundcore shpchp mac_hid parport_pc ppdev lp parport autofs4 hid_apple squashfs overlay nls_utf8 isofs nls_iso8859_1 jfs xfs libcrc32c reiserfs dm_mirror dm_region_hash dm_log uas usb_storage hid_generic usbhid hid amdkfd amd_iommu_v2 radeon firewire_ohci psmouse i2c_algo_bit ttm firewire_core drm_kms_helper crc_itu_t ahci libahci wmi e1000e ptp drm pps_core pata_acpi [last unloaded: hid_apple]
[  +0.000042] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W       4.2.0-16-generic #19-Ubuntu
[  +0.000001] Hardware name: Dell Inc. OptiPlex 780                 /0C27VV, BIOS A15 08/06/2013
[  +0.000003]  0000000000000000 c856fe507f31ed24 ffff8802b7c03b38 ffffffff817e8c09
[  +0.000002]  0000000000000000 ffff8802b7c03b90 ffff8802b7c03b78 ffffffff8107b3c6
[  +0.000003]  0000000000000000 ffff88027e2f8300 0000000000000003 ffff880293890000
[  +0.000002] Call Trace:
[  +0.000001]  <IRQ>  [<ffffffff817e8c09>] dump_stack+0x45/0x57
[  +0.000005]  [<ffffffff8107b3c6>] warn_slowpath_common+0x86/0xc0
[  +0.000003]  [<ffffffff8107b455>] warn_slowpath_fmt+0x55/0x70
[  +0.000002]  [<ffffffff815df312>] usb_submit_urb.part.6+0x142/0x550
[  +0.000002]  [<ffffffff815df782>] usb_submit_urb+0x62/0x70
[  +0.000004]  [<ffffffffc054420b>] acm_submit_read_urb+0x3b/0x80 [cdc_acm]
[  +0.000003]  [<ffffffffc0544e39>] acm_read_bulk_callback+0xd9/0x170 [cdc_acm]
[  +0.000002]  [<ffffffff815dbf75>] __usb_hcd_giveback_urb+0x85/0x130
[  +0.000002]  [<ffffffff815dc05d>] usb_hcd_giveback_urb+0x3d/0xf0
[  +0.000002]  [<ffffffff81614771>] uhci_giveback_urb+0xb1/0x290
[  +0.000003]  [<ffffffff811cc6cf>] ? dma_pool_free+0x9f/0xc0
[  +0.000002]  [<ffffffff81616537>] uhci_scan_schedule.part.31+0x4b7/0xbd0
[  +0.000003]  [<ffffffff81616ffe>] uhci_irq+0xde/0x190
[  +0.000002]  [<ffffffff815db636>] usb_hcd_irq+0x26/0x40
[  +0.000002]  [<ffffffff810d41d4>] handle_irq_event_percpu+0x74/0x180
[  +0.000002]  [<ffffffff810d4329>] handle_irq_event+0x49/0x70
[  +0.000002]  [<ffffffff810d737a>] handle_fasteoi_irq+0x9a/0x150
[  +0.000002]  [<ffffffff81017295>] handle_irq+0x25/0x40
[  +0.000003]  [<ffffffff817f25ef>] do_IRQ+0x4f/0xe0
[  +0.000002]  [<ffffffff817f056b>] common_interrupt+0x6b/0x6b
[  +0.000001]  <EOI>  [<ffffffff8101f3ac>] ? mwait_idle+0x8c/0x140
[  +0.000005]  [<ffffffff8101f94f>] arch_cpu_idle+0xf/0x20
[  +0.000002]  [<ffffffff810bd3ea>] default_idle_call+0x2a/0x40
[  +0.000002]  [<ffffffff810bd729>] cpu_startup_entry+0x2c9/0x320
[  +0.000003]  [<ffffffff817dd16c>] rest_init+0x7c/0x80
[  +0.000004]  [<ffffffff81d50025>] start_kernel+0x48b/0x4ac
[  +0.000002]  [<ffffffff81d4f120>] ? early_idt_handler_array+0x120/0x120
[  +0.000002]  [<ffffffff81d4f339>] x86_64_start_reservations+0x2a/0x2c
[  +0.000002]  [<ffffffff81d4f485>] x86_64_start_kernel+0x14a/0x16d
[  +0.000002] ---[ end trace 7d80c581b767159f ]---

me@host:~$ sudo lsusb -v -s10:6

Bus 010 Device 006: ID 16d0:087e MCS 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            2 Communications
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x16d0 MCS
  idProduct          0x087e 
  bcdDevice            1.00
  iManufacturer           1 digistump.com
  iProduct                2 Digispark Serial
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           67
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              0 
      CDC Header:
        bcdCDC               1.10
      CDC ACM:
        bmCapabilities       0x02
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      CDC Call Management:
        bmCapabilities       0x03
          call management
          use DataInterface
        bDataInterface          1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval             255
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)
```

jayeye, 2016-01-28:
I'm having the same problem.

quigleymd, 2016-05-04:
I've got the same problem on Fedora 23, any luck resolving this?

`[quigley@mythbox ~]$ uname -a
Linux mythbox.localdomain 4.2.3-300.fc23.x86_64 #1 SMP Mon Oct 5 15:42:54 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux`

vshymanskyy, 2016-05-04:
+1, same issue

paulfantom, 2016-06-05:
+1, Same thing on openSUSE Leap 42.1

uname -a
Linux chameleon 4.1.21-14-default #1 SMP PREEMPT Sun Apr 17 07:27:45 UTC 2016 (fc187c1) x86_64 x86_64 x86_64 GNU/Linux

shing19m, 2016-08-13:
+1 
uname -a
Linux desktop 3.16.0-4-amd64 #1 SMP Debian 3.16.7-ckt25-2 (2016-04-08) x86_64 GNU/Linux

probonopd, 2016-08-13:
Maybe someone can use this technique to debug this: https://m.youtube.com/watch?v=-NxoNdTj_7U

paulfantom, 2016-08-20:
I think I found what is an issue here. DigisparkCDC is using three endpoints and bulk transfer on low-speed device, which violates USB protocol (via http://www.recursion.jp/prose/avrcdc/).

probonopd, 2016-08-20:
Thanks @paulfantom can you propose a fix?

paulfantom, 2016-08-20:
I was searching for a fix, but I came with nothing. Also programming USB shenanigans is way out of my knowledge.

harvald, 2016-08-29:
I think i found solution.

DigiCDC is configured to create 1 interrupt endpoint, and 2 bulk endpoints. Bulk endpoints are forbidden for low speed usb (1,5Mbit/s). So this is a bug. Until now Linux kernel was forgiving, because some older hardware had the same bug in theirs firmware. 

It seems it's enough to change two exact same lines in DigiCDC.cpp.
from `0x02,        /* attrib: Bulk endpoint */ ` 
to `0x03,        /* attrib: Interrupt endpoint */ ` 

I don't know if this change is sufficient for full compliance with USB standard, but i have no more errors in linux logs :) Unfortunately there is possibility, that windows drivers should be also altered. I didn't tested that yet.

probonopd, 2016-08-29:
Where do I find DigiCDC.cpp? I can't find it using the search for this repo.

harvald, 2016-08-29:
DigiCDC.cpp is part of DigiCDC library and it is installed somewhere in Arduino's directories. On windows xp i have it in: C:\Documents and Settings\My user\Local settings\appdata\Arduino15\packages\digistump\hardware\avr\1.6.7\libraries\DigisparkCDC\

probonopd, 2016-08-29:
On which system is this working for you? On Ubuntu 16.04 I get immediate kernel module crashes in dmesg when I plug the device in and it loads the user program.

harvald, 2016-08-29:
It seems to work on Ubuntu 16.04.1 desktop livecd. On ubuntu server 16.04.1 with newest kernel don't work either. I will test this further tomorrow.

probonopd, 2016-11-12:
@cnlohr to the rescue ;-)

cnlohr, 2016-11-23:
What?  Not sure what I'm supposed to be doing here :-p

AFAIK you aren't "allowed" to do a CDC device over low speed.