digistump/DigistumpArduino #17: DigiKeyboard.h example doesn't compile

dchapkine, 2016-01-10:
I am using digispark tiny board on ubuntu 14.04, with platformio.

If i include "DigiKeyboard.h" my project doesn't compile anymore

Steps to reproduce this issue using platformio:

```
mkdir tmp
cd tmp
platformio init --board digispark-tiny
```

src/main.ino
```
#include "DigiKeyboard.h"

void setup() {
  // don't need to set anything up to use DigiKeyboard
}

void loop() {
  DigiKeyboard.sendKeyStroke(0);
  DigiKeyboard.println("Hello Digispark!");
  DigiKeyboard.delay(5000);
}

```

Error output
```
19:22 $ platformio run
[Sun Jan 10 19:22:38 2016] Processing digispark-tiny (platform: atmelavr, board: digispark-tiny, framework: arduino)
--------------------------------------------------------------------------------------------------------------------------

scons: *** Multiple ways to build the same target were specified for: /home/dchapkine/Desktop/digispark-tiny/tmp/.pioenvs/digispark-tiny/DigisparkKeyboard/usbdrvasm.o  (from ['/home/dchapkine/Desktop/digispark-tiny/tmp/.pioenvs/digispark-tiny/DigisparkKeyboard/usbdrvasm.S'] and from ['/home/dchapkine/Desktop/digispark-tiny/tmp/.pioenvs/digispark-tiny/DigisparkKeyboard/usbdrvasm.asm'])
File "/usr/local/lib/python2.7/dist-packages/platformio/builder/tools/platformio.py", line 201, in BuildLibrary
============================================== [ ERROR ] Took 0.20 seconds ==============================================

```

Note that if i am using any other example, like the mouse example, it works:
```
#include <DigiMouse.h>

void setup() {
  DigiMouse.begin();
}

void loop() {
  DigiMouse.moveY(-20); //up 20
  DigiMouse.delay(500);
  DigiMouse.moveX(20); //right 20
  DigiMouse.delay(500);
  DigiMouse.moveY(20); // down 20
  DigiMouse.delay(500);
  DigiMouse.moveX(-20); //left 20
  DigiMouse.delay(500);
}

```

But as soon as i include DigiKeyboard, it refuses to compile with the error above

stevemcquaid, 2016-08-31:
@dchapkine - Had the same issue. Was able to solve it by compiling the library myself + removing the DigisparkKeyboard/usbdrvasm.asm file.

The error is caused becuase usbdrvasm.asm includes usbdrvasm.S ... Not sure why, but if both of them declare the same thing, it shows that error.

TL;DR; Remove usbdrvasm.asm file.

crazyquark, 2017-02-03:
I am also having this issue

ardentsword, 2017-03-03:
Having the same issue here with the latest version of PlatformIO (core v3.2.1). Deleting usbdrvasm.asm solved this problem for me.

Slncio, 2018-10-07:
Check you have the digispark as the default board in arduion.