digistump/DigistumpArduino #20: IRhashdecode.ino: IRdecodeHash does not name a type

probonopd, 2016-01-24:
Arduino-1.6.8.201601190826

```
IRhashdecode:24: error: 'IRdecodeHash' does not name a type
 IRdecodeHash My_Hash_Decoder;
 ^
/root/.arduino15/packages/digistump/hardware/avr/1.6.5/libraries/DigisparkIRLib/examples/IRhashdecode/IRhashdecode.ino: In function 'void loop()':
IRhashdecode:37: error: 'My_Hash_Decoder' was not declared in this scope
     My_Hash_Decoder.copyBuf(&My_Decoder);//copy the results to the hash decoder
     ^
exit status 1
'IRdecodeHash' does not name a type
```

digistump, 2016-02-13:
IRdecodeHash only works if you set your protocol to hash in IRLib.h (in the DigisparkIRLib libary folder)

closed 2016-02-13 by digistump

probonopd, 2016-02-13:
Thanks @digistump. Would be nice if this was wrapped in a way that throws a more verbose error then. ;-)