digistump/DigistumpArduino #22: Missing examples issue in Arduino IDE 1.6.7

GertSanders, 2016-01-25:
Hi, I added the board definition URL for the board manager and I can get the definitions of the Digistup boards. But when I try to compile I get this warning:

Invalid library found in /Users/gertsanders/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/Digispark_Examples

My Arduino IDE is version 1.6.7 (so not 1.6.5). I also do not see any board when connecting the USB cable, probably again because the board manager link is for the older IDE.

Is there an update for Arduino.cc IDE 1.6.7 ?

I'm using a Mac on OSX 10.11.2

rac2030, 2016-01-31:
I get the same message (2 times) on OSX 10.10.4 with IDE 1.6.7
But upload runs fine.
Have it unplugged, hit upload and plug it in once you see the message from the digispark uploader that you can plug it in now.
Example of what the terminal looks like (beyond the sketch itself)
`Running Digispark Uploader...
Plug in device now... (will timeout in 60 seconds)
> Please plug in the device ... 
> Press CTRL+C to terminate the program.
> Device is found!
connecting: 16% complete`

per1234, 2016-02-04:
I have submitted a fix for the missing examples and `Invalid library...` warnings issue at: https://github.com/digistump/DigistumpArduino/issues/28

FlackMonkey, 2016-02-05:
Same thing with usb digispark(default 16.5mhz) attiny85 on windows XP and Arduino IDE 1.6.7

> WARNING: Category '' in library SPI is not valid. Setting to 'Uncategorized'
Warning: platform.txt from core 'Digistump AVR Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
Warning: platform.txt from core 'Digistump AVR Boards' contains deprecated recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mmcu={build.mcu} -o "{build.path}/{build.project_name}.elf" {object_files} "{build.path}/{archive_file}" "-L{build.path}" -lm, automatically converted to recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mmcu={build.mcu} -o "{build.path}/{build.project_name}.elf" {object_files} "{archive_file_path}" "-L{build.path}" -lm. Consider upgrading this core.

Sketch uses 668 bytes (11%) of program storage space. Maximum is 6,012 bytes.
Global variables use 9 bytes of dynamic memory.
Running Digispark Uploader...
Plug in device now... (will timeout in 60 seconds)
Assertion failed: res >= 4, file library/micronucleus_lib.c, line 100

This application has requested the Runtime to terminate it in an unusual way.
Please contact the application's support team for more information.
Invalid library found in C:\Documents and Settings\ADMIN\Local Settings\Application Data\Arduino15\packages\digistump\hardware\avr\1.6.5\libraries\Digispark_Examples: C:\Documents and Settings\ADMIN\Local Settings\Application Data\Arduino15\packages\digistump\hardware\avr\1.6.5\libraries\Digispark_Examples
Invalid library found in C:\Documents and Settings\ADMIN\Local Settings\Application Data\Arduino15\packages\digistump\hardware\avr\1.6.5\libraries\Digispark_Examples: C:\Documents and Settings\ADMIN\Local Settings\Application Data\Arduino15\packages\digistump\hardware\avr\1.6.5\libraries\Digispark_Examples
Invalid library found in C:\Documents and Settings\ADMIN\Local Settings\Application Data\Arduino15\packages\digistump\hardware\avr\1.6.5\libraries\Digispark_Examples: C:\Documents and Settings\ADMIN\Local Settings\Application Data\Arduino15\packages\digistump\hardware\avr\1.6.5\libraries\Digispark_Examples

closed 2016-02-12 by digistump

Tares8, 2022-07-15:
Although my board is connected I get this message:
> Please plug in the device ... 
> Press CTRL+C to terminate the program.
> Device search timed out
HELPPPPP