digistump/DigistumpArduino #26: SPI compile issues on Arduino IDE 1.6.7

rac2030, 2016-02-02:
There are compile issues if SPI.h get's included (directly or over a library).
I noticed first a warning at the beginning
`WARNING: Category '' in library SPI is not valid. Setting to 'Uncategorized'`
The following is a example output from rf24ping85 sample code but the same happens with other code as well. For reproduction it is enough to create a empty sketch and put `#include "SPI.h"` at the top.
On the same machine (using OSX) a parallel run of Arduino IDE 1.6.5 (Sharing libs and settings) the same sketch compiles fine.

```
In file included from 
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/avr/io.h:99:0,
                 from /Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/avr/interrupt.h:38,
                 from /users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/cores/tiny/WProgram.h:8,
                 from /users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/cores/tiny/Arduino.h:4,
                 from /users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.h:17,
                 from /users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.cpp:14:
/users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.h: In member function 'void SPISettings::init_AlwaysInline(uint32_t, uint8_t, uint8_t)':
/users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.h:146:16: error: 'SPE' was not declared in this scope
     spcr = _BV(SPE) | _BV(MSTR) | ((bitOrder == LSBFIRST) ? _BV(DORD) : 0) |
                ^
/users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.h:146:27: error: 'MSTR' was not declared in this scope
     spcr = _BV(SPE) | _BV(MSTR) | ((bitOrder == LSBFIRST) ? _BV(DORD) : 0) |
                           ^
/users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.h:146:65: error: 'DORD' was not declared in this scope
     spcr = _BV(SPE) | _BV(MSTR) | ((bitOrder == LSBFIRST) ? _BV(DORD) : 0) |
                                                                 ^
In file included from /users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.cpp:14:0:
/users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.h: In static member function 'static void SPIClass::beginTransaction(SPISettings)':
/users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.h:203:5: error: 'SPCR' was not declared in this scope
     SPCR = settings.spcr;
     ^
/users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.h:204:5: error: 'SPSR' was not declared in this scope
     SPSR = settings.spsr;
     ^
/users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.h: In static member function 'static uint8_t SPIClass::transfer(uint8_t)':
/users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.h:209:5: error: 'SPDR' was not declared in this scope
     SPDR = data;
     ^
/users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.h:217:14: error: 'SPSR' was not declared in this scope
     while (!(SPSR & _BV(SPIF))) ; // wait
              ^
In file included from /Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/avr/io.h:99:0,
                 from /Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/avr/interrupt.h:38,
                 from /users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/cores/tiny/WProgram.h:8,
                 from /users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/cores/tiny/Arduino.h:4,
                 from /users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.h:17,
                 from /users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.cpp:14:
/users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.h:217:25: error: 'SPIF' was not declared in this scope
     while (!(SPSR & _BV(SPIF))) ; // wait
                         ^
In file included from /users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.cpp:14:0:
/users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.h: In static member function 'static uint16_t SPIClass::transfer16(uint16_t)':
/users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.h:223:11: error: 'SPCR' was not declared in this scope
     if (!(SPCR & _BV(DORD))) {
           ^
In file included from /Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/avr/io.h:99:0,
                 from /Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/avr/interrupt.h:38,
                 from /users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/cores/tiny/WProgram.h:8,
                 from /users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/cores/tiny/Arduino.h:4,
                 from /users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.h:17,
                 from /users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.cpp:14:
/users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.h:223:22: error: 'DORD' was not declared in this scope
     if (!(SPCR & _BV(DORD))) {
                      ^
In file included from /users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.cpp:14:0:
/users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.h:224:7: error: 'SPDR' was not declared in this scope
       SPDR = in.msb;
       ^
/users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.h:226:16: error: 'SPSR' was not declared in this scope
       while (!(SPSR & _BV(SPIF))) ;
                ^
In file included from /Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/avr/io.h:99:0,
[... 187 lines cut ...]
                 ^
/users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.cpp:59:13: error: 'SCK' was not declared in this scope
     pinMode(SCK, OUTPUT);
             ^
/users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.cpp:60:13: error: 'MOSI' was not declared in this scope
     pinMode(MOSI, OUTPUT);
             ^
/users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.cpp: In static member function 'static void SPIClass::end()':
/users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.cpp:74:5: error: 'SPCR' was not declared in this scope
     SPCR &= ~_BV(SPE);
     ^
In file included from /Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/avr/io.h:99:0,
                 from /Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/avr/interrupt.h:38,
                 from /users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/cores/tiny/WProgram.h:8,
                 from /users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/cores/tiny/Arduino.h:4,
                 from /users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.h:17,
                 from /users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.cpp:14:
/users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI/SPI.cpp:74:18: error: 'SPE' was not declared in this scope
     SPCR &= ~_BV(SPE);
                  ^
Using library RF24 at version 1.1.5 in folder: /users/**/Documents/Arduino/libraries/RF24 
Using library SPI at version 1.0 in folder: /users/**/Library/Arduino15/packages/digistump/hardware/avr/1.6.5/libraries/SPI 
exit status 1
Error compiling.
```

digistump, 2016-02-13:
Fixed in newest release coming in a few minutes - thanks for the report!

closed 2016-02-13 by digistump

Ole2mail, 2016-09-04:
while using version Arduino IDE 1.6.11 the issue appear to be similar:
In file included from /Applications/Arduino-CH340-recovery.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h:28:0,
                 from /Applications/Arduino-CH340-recovery.app/Contents/Java/hardware/arduino/avr/libraries/SPI/src/SPI.h:17,
                 from /Applications/Arduino-CH340-recovery.app/Contents/Java/hardware/arduino/avr/libraries/SPI/src/SPI.cpp:14:
/Applications/Arduino-CH340-recovery.app/Contents/Java/hardware/arduino/avr/libraries/SPI/src/SPI.cpp: In static member function 'static void SPIClass::begin()':
/Applications/Arduino-CH340-recovery.app/Contents/Java/hardware/arduino/avr/libraries/SPI/src/SPI.cpp:32:37: error: 'SS' was not declared in this scope
     uint8_t port = digitalPinToPort(SS);
                                     ^
/Applications/Arduino-CH340-recovery.app/Contents/Java/hardware/arduino/avr/libraries/SPI/src/SPI.cpp:32:20: note: in expansion of macro 'digitalPinToPort'
     uint8_t port = digitalPinToPort(SS);
                    ^
/Applications/Arduino-CH340-recovery.app/Contents/Java/hardware/arduino/avr/libraries/SPI/src/SPI.cpp:33:39: error: 'SS' was not declared in this scope
     uint8_t bit = digitalPinToBitMask(SS);
                                       ^
/Applications/Arduino-CH340-recovery.app/Contents/Java/hardware/arduino/avr/libraries/SPI/src/SPI.cpp:33:19: note: in expansion of macro 'digitalPinToBitMask'
     uint8_t bit = digitalPinToBitMask(SS);
                   ^
/Applications/Arduino-CH340-recovery.app/Contents/Java/hardware/arduino/avr/libraries/SPI/src/SPI.cpp:39:20: error: 'SS' was not declared in this scope
       digitalWrite(SS, HIGH);
                    ^
/Applications/Arduino-CH340-recovery.app/Contents/Java/hardware/arduino/avr/libraries/SPI/src/SPI.cpp:45:13: error: 'SS' was not declared in this scope
     pinMode(SS, OUTPUT);
             ^
/Applications/Arduino-CH340-recovery.app/Contents/Java/hardware/arduino/avr/libraries/SPI/src/SPI.cpp:59:13: error: 'SCK' was not declared in this scope
     pinMode(SCK, OUTPUT);
             ^
/Applications/Arduino-CH340-recovery.app/Contents/Java/hardware/arduino/avr/libraries/SPI/src/SPI.cpp:60:13: error: 'MOSI' was not declared in this scope
     pinMode(MOSI, OUTPUT);
             ^
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.