digistump/DigistumpArduino #29: CRC doesn't match. File is corrupted.

DataPaulette, 2016-02-15:
When I try to install your hardware with the Arduino Boardmanager I get the following ERROR :

CRC doesn't match. File is corrupted.
java.lang.RuntimeException: java.lang.Exception: CRC doesn't match. File is corrupted.
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$24(ContributionManagerUI.java:170)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.Exception: CRC doesn't match. File is corrupted.
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:77)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:110)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$24(ContributionManagerUI.java:164)
	... 1 more

WayneBuckhanan, 2016-02-16:
The json file at http://digistump.com/package_digistump_index.json is missing a digit in the SHA-256.

Is:
7F7E9F5AB982163F7A792C411326F0192A35B8A90890B793F2F424A5426583D
Should be:
7f7e9f5ab982163f7a792c411326f0192a35b8a90890b793 **4** f2f424a5426583d

Please let us know when that is updated.
Thanks!

alebarra, 2016-02-16:
the error happened to me too, but sha256 checksum is correct and no digit is missing :
sha256 file:
7f7e9f5ab982163f7a792c411326f0192a35b8a90890b7934f2f424a5426583d  digistump-avr-1.6.7.zip
package_digistump_index.json string:
          "checksum": "SHA-256:7F7E9F5AB982163F7A792C411326F0192A35B8A90890B7934F2F424A5426583D",

alebarra, 2016-02-17:
[SOLVED]
the issue is resolved for me by deleting ALL file in $HOME/.arduino15/staging/packages/ , seems the error is caused by some partially downloaded pakage.

closed 2016-04-08 by digistump

mrohner, 2017-03-02:
I do not have such a directory on Windows 10. The error is still here