digistump/DigistumpArduino #33: Digikeyboard needs implementation of NumLock/CapsLock/ScrollLock status

Akasurde, 2016-04-16:
As title suggest, in one of my project I need to detect status of Caps Lock button on host machine in digispark. 

Currently, DigiKeyboard library doesn't implement such functionality. I am newbie in USB stuff so not aware how to implement such functionality.

I think it is something related to "USBRQ_HID_SET_REPORT"

felixmeyer6, 2018-01-01:
Hey.
To enable/disable caps lock, just type into your code : 
 DigiKeyboard.sendKeyStroke(57);

Hope it helps !

Akasurde, 2018-01-02:
I need to get status of NumLock/CapsLock/ScrollLock keys from USB HOST to digispark. My use case is I want to perform some operation on DigiSpark (like blink LED) on USB host presses any of above keys.

felixmeyer6, 2018-01-02:
Oh, okay, I dunno that. Sorry for misunderstanding. But I think an HID
device can't get some info on the computer...

Semnodime, 2020-01-25:
> I think it is something related to "USBRQ_HID_SET_REPORT"

Via `usbHidReportDescriptor` [`Digikeyboard.h`](digistump-avr/libraries/DigisparkKeyboard/DigiKeyboard.h) tells the host to not send reports for status LED updates.

This is probably done due to the limitations of the currently used USB driver `usbdrv.h`.

More info on the `usbdrv.h`
```
/* Name: usbdrv.h
 * Project: V-USB, virtual USB port for Atmel's(r) AVR(r) microcontrollers
 * Author: Christian Starkjohann
 * Creation Date: 2004-12-29
 * Tabsize: 4
 * Copyright: (c) 2005 by OBJECTIVE DEVELOPMENT Software GmbH
 * License: GNU GPL v2 (see License.txt), GNU GPL v3 or proprietary (CommercialLicense.txt)
 * This Revision: $Id: usbdrv.h 793 2010-07-15 15:58:11Z cs $
 */
```