digistump/DigistumpArduino #34: Unable to Find EEPROM.h Include

prologician, 2016-05-15:
Environment: Win7 x64; Arduino IDE 1.6.7; Digistump AVR Boards v1.6.7
Board: Digispark (Default - 16.5mhz)

From the specification sheet http://www.atmel.com/devices/attiny85.aspx, ATTiny85 has 512 bytes of EEPROM memory; the forum thread https://digistump.com/board/index.php?topic=1132.0 shows that this is accessible using the functions provided through the EEPROM.h include. However, it appears that that include cannot be found during the build process.

Sample program (blank program, plus the include):
```
#include "EEPROM.h"

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```
Error produced:
```
fatal error: EEPROM.h: No such file or directory

 #include <EEPROM.h>

                    ^

compilation terminated.

exit status 1
Error compiling.
```

digistump, 2016-05-17:
It looks like the Arduino team moved this library out of the global scope in recent releases - you can grab and use this copy here: [EEPROM.zip](https://github.com/digistump/DigistumpArduino/files/267174/EEPROM.zip)

It will also be fixed the next time we push a release.

Thanks

closed 2016-05-17 by digistump

LightningStalker, 2019-10-07:
It didn't get fixed

bratoff, 2019-11-05:
Your include is incorrect.  AVR-specific header files are in the avr subfolder, so the statement should be:

#include <avr/eeprom.h>

Make that change and your test code compiles.

marko-pi, 2021-05-28:
Could you be more specific about which folder you put eeprom.h in? In my case, eeprom.h is called from the NmraDCC library and I don't want to mess up the library.

BTW, is it possible that the Micronucleus bootloader and NmraDCC conflict since NmraDCC accesses the EEPROM where the Micronucleus is located?

pfeerick, 2021-05-28:
You *should* be able to install it directly via the Arduino IDE
[image]

If that doesn't work for some reason, this is where it is in the repo structure... so depending on your OS, you'll need to golooking for the `Arduino15\packages` directory, and then you should be able to figure it out from there. On Windows, this is located in `%localappdata%` and I think it's the home (~) directory on Mac and Linux. 

https://github.com/digistump/DigistumpArduino/tree/master/digistump-avr/libraries/EEPROM

I would strongly suggest you look at https://github.com/ArminJo/DigistumpArduino also as a more up to date and actively supported version of the Arduino core for these boards ;)

marko-pi, 2021-05-28:
Thanks, it worked.  Plus I am using the latest hex version of Digistump.

NmraDCC is unfortunatelly not working.  Could be the frequency difference as this library heavily depends on precise timing.  (8MHz vs 16.5MHz)