digistump/DigistumpArduino #37: DPinst64.exe fails to install Digistump LLC (usbser) Ports (08/162014 1.1.0.0)

BubbleBoyo, 2016-06-30:
DPinst64.exe fails to install Digistump LLC (usbser) Ports (08/162014 1.1.0.0) | Install failed

Using DigiSerial.inf fails:
Install Error:
"The hash for the file is not present in the specified catalog file. The file is Likely corrupt or the victim of tampering."

Get-FileHash .\Digistump.Drivers.zip -Algorithm SHA1
A1D36DF70125BF1BDCD579BFCE64105BC1728F4D

Get-FileHash .\digiserial.cat -Algorithm SHA1
74D50FFB989DE07CCA8028899743863E4F659DCD

Get-FileHash .\DigiSerial.inf -Algorithm SHA1
69C3409E9D6D13B15A7962F92036EFCA30DC9C2C

digiserial.cat file lists the sha1 for DigiSerial.inf as being:
4DA6FDC07A46FA211FFF1D2621FB5C4AECCF361F

The hashes don't match.

rumlyne, 2016-07-05:
I too cannot install the usbser driver.

tatobari, 2016-07-27:
Me neither.

tatobari, 2016-08-13:
The problem is most simple. According to [this post in DigiStump's Forum](https://digistump.com/board/index.php?topic=1637.0), the INF file had a mistake and was fixed. It lacked the line `usbser.sys` below the line that says `[DriverInstall.NTamd64.CopyFiles]`.

The current one has that line. I removed the line and ran `Get-FileHash .\digiserial.inf -Algorithm SHA1`. I got `4DA6FDC07A46FA211FFF1D2621FB5C4AECCF361F` just like @BubbleBoyo said the CAT file lists.

I'm not expert but, as far as I got here, I guess the CAT file needs to be rebuilt to list the HASH `69C3409E9D6D13B15A7962F92036EFCA30DC9C2C`, which is the INF's file hash when it includes the line  `usbser.sys` below the line that says `[DriverInstall.NTamd64.CopyFiles]`.

treii28, 2016-08-14:
+1

Multiple threads on the digistump forums with many echos. No responses or answers.

olegoleg1111121, 2021-02-20:
+

ArminJo, 2021-02-20:
Use https://github.com/ArminJo/DigistumpArduino and you will be lucky 😀

Zedasix, 2023-12-01:
Same problem

pfeerick, 2023-12-11:
@Zedasix Did you try the https://github.com/ArminJo/DigistumpArduino BSP to see if it has been resolved there? This BSP is basically abandoned, the last update was six years ago.