digistump/DigistumpArduino #38: IRsendDemo and other uses of IRLib.h fail to compile

GeReV, 2016-07-24:
I've been trying to compile some code using the Digispark IRLib, but so far have been getting the following error message.

Got this same one with the AVR libraries 1.6.7 and 1.6.5A, in IDE versions 1.6.9 and 1.6.5-r5.

```
IRsendDemo:11: error: 'IRsend' does not name a type

 IRsend My_Sender;

 ^

C:\Users\Me\Documents\Arduino\libraries\DigisparkIRLib\examples\IRsendDemo\IRsendDemo.ino: In function 'void loop()':

IRsendDemo:22: error: 'My_Sender' was not declared in this scope

     My_Sender.send(SONY,0xa8bca, 20);

     ^

Multiple libraries were found for "IRLib.h"
 Used: C:\Users\Me\Documents\Arduino\libraries\DigisparkIRLib
 Not used: C:\Users\Me\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\libraries\DigisparkIRLib
exit status 1
'IRsend' does not name a type
```

digistump, 2016-07-26:
It appears you have a conflicting library at C:\Users\Me\Documents\Arduino\libraries\DigisparkIRLib - can you try removing that and see if the issue presists?

closed 2016-07-26 by digistump

GeReV, 2016-07-26:
@digistump, I did remove it, eventually I discovered that I was missing the `USE_IR_SEND` define and `IR_USE_TIMER1` under `IRLibTimer.h`.
I could compile my code and use `IRsendRaw.send`.

Seems that its default behavior is apparently to define `IR_USE_TIMER2` since it falls back to the `#else` branch.
All other options would fail due to different variables/constants not being defined (i.e. TIMSK).
I can provide more logs on that at a later time.

But when it did compile, none of the pins on the board (Digispark Pro) didn't seem to send the signal.

I admit this is a bit over my head, as I'm new to the Arduino world, but according to [IRremote's Wiki](https://github.com/z3t0/Arduino-IRremote/wiki), ATtiny85 boards use the TINY0 timer and not TIMER1, could this be related?