digistump/DigistumpArduino #41: DigiCDC does not work with macOS Sierra

amckinlay, 2016-10-10:
After uploading the sketch from `File → Examples → DigiCDC → Print`, reconnecting the original DigiSpark, and attempting to attach the Arduino Serial Monitor, the following error message is printed in the Arduino error window:
```
Error opening serial port '/dev/cu.usbmodem1421'. (Port busy)
```

Looking at the device tree:
```
▶ ls /dev/*usbmodem*
/dev/cu.usbmodem1421  /dev/tty.usbmodem1421
```

But `lsof | grep "/dev/cu.usbmodem1421"` outputs nothing.

How the device appears in System Information:
[image]

# Version info

Arduino 1.6.12

```
▶ sw_vers
ProductName:	Mac OS X
ProductVersion:	10.12
BuildVersion:	16A323
```

```
▶ sysctl hw.model
hw.model: MacBookPro12,1
```

DeuxVis, 2016-10-11:
You might want to post that on the [digistump forums](http://digistump.com/board/), more people will be able to help you there.
I believe github is more for bug reports / code contributions on the software than help requests on using the product.

amckinlay, 2016-10-12:
@DeuxVis Seems like the issue tracker and forums have been abandoned anyway.

DeuxVis, 2016-10-12:
Well you got a point, drawing attention is more likely to succeed here.

gemby, 2016-10-13:
It does not work under El Capitan either, same symptoms, port busy

`Arduino: 1.6.8 (Mac OS X), Board: "Digispark (Default - 16.5mhz)"

Error opening serial port '/dev/cu.usbmodem14131'. (Port busy)

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

ikeji, 2016-12-09:
+1 it doesn't work for my ElCapitan.

chappy1978, 2017-02-25:
It isnt working on MBP with Sierra same errors and everything. Been through the Digistump forums and there having the same problems. Worked threw several different solution I found online and still nothing. I got this from the Console when I try to open the DigiCDC in Arduino serial. 
002732.611858 StandardUSB::validateEndpointMaxPacketSize: USB 2.0 5.[5|7].3: endpoint 0x81 invalid wMaxPacketSize 0x0008
Does that mean anything?

chappy1978, 2017-02-25:
Cool!!! I found out the invalid wMaxPacketSize 0x0008 is directly related to the DigiCDC.h library.
#define HW_CDC_BULK_OUT_SIZE     8
#define HW_CDC_BULK_IN_SIZE      8

When I change these two line I get different wMaxPacketSize in the Console. Does anyone know what this means?

chappy1978, 2017-02-25:
Correction its the #define HW_CDC_BULK_IN_SIZE      8 line int he DigiCDC.h that it effects

amckinlay, 2017-02-25:
Might have something to do with the fact that the USB emulation library for
AT* cores is not standards compliant, and now MacOS has decided not to
allow an invalid configuration.

chappy1978, 2017-02-26:
Update I found the source of the error, but I still have no USBSerial. The error comes from line 290 in DigiCDC.cpp.  
0x81,        /* IN endpoint number 1 */
Once I changed 0x81 to 0x01 the error "StandardUSB::validateEndpointMaxPacketSize: USB 2.0 5.[5|7].3: endpoint 0x81 invalid wMaxPacketSize 0x0008" goes away. But still no SerialUSB. Any help?

chappy1978, 2017-02-26:
After of couple of days of searching and studying. I put on a logic analyzer on the USB bus of the Digispark. When I upload a sketch from on Arduino USB. The Digispark is unplugged and plugged back in, at that time the OS gives it an address to identify the board on the USB bus. Then it uploads the sketch to the board.
   What I found is that the MacOS is first giving the digispark an address on EP0 and then when you try to open a serial terminal it is making a call to a totally different address on EP3. Thus the digispark doesn't know to answer to the call. The port isn't busy there is sadly a big problem with the new IOUSBHostFamily that Apple change in the new revision of macOS.

Defragster, 2017-02-27:
Do you have a PC (Linux/Windows) to try the same 'analysis'? Not sure about the Digispark - but often a device will show differently to the OS between program/bootloader and run times and get a different driver association and even port assignment. As long as both are working and sorted right on the OS side the device typically can work if it is running appropriately on its end.

chappy1978, 2017-02-27:
No I do not have a Win/Linux handy. What I am trying to describe here is a MacOS Sierra only problem. The Digispark works on old MacOS but it seems Apple redesigned the way that USB Driver behaves in Sierra. It seems they do not much care for USB 1.1 these days. I am sure the Digispark is working just fine. MacOS on the USB protocol level is calling for the wrong address when its looking for a serial device only. So the Digispark does not realize that it is being call for.  New sketches can be uploaded just fine on Sierra, just the DigiCDC can not be used until Apple fixes the problem. I have sent my information to Apple maybe they just might fix. lol

Jeff

Defragster, 2017-02-27:
I expected it to show the same but working - just to understand the mechanism . . . and show it was the computer at fault. Try with (or without ?) a hub just on the chance that can change something?

chappy1978, 2017-02-27:
I gave it a shot before with no luck but I did it again. I only have a 2.0 hub handy. It still didnt work.