digistump/DigistumpArduino #45: yield() is missing for the AVR boards

nilrog, 2016-12-16:
It is not possible to compile a sketch that is using the latest FastLED library for the original Digispark boards.

```
FastLED.cpp:133:9: error: 'yield' was not declared in this scope

   yield();

         ^

Using library FastLED at version 3.1.3
```

Arduino IDE v1.6.13....and Digistump AVR Boards v1.6.7.

pfeerick, 2017-04-08:
As far as I can tell, this can be fixed merely with addition of the hooks.c file that is part of the standard Arduino IDE core to the Digispark core... it implements a yield() that does nothing as it basically doesn't do anything on AVR microcontrollers, thus letting the code compile.

Nzbuu, 2017-12-23:
Is there any progress on this?

pfeerick, 2017-12-24:
@Nzbuu I don't think so, not much has been happening in Digispark land of late. Your best bet is to follow through to the pull request above, and patch your own files. It's pretty simple. just copy the hooks.c file into  one (or both core folders if you use both the Digispark and Digispark Pro), and then add the ```void yield(void);``` to the ```Arduino.h``` and/or ```WProgram.h``` file in the core folder. It's all there in the PR.