digistump/DigistumpArduino #46: scancode-ascii-table.h: Support for QWERTZ and AZERTY keyboards.

likeguest, 2016-12-23:
By now, scancode-ascii-table.h is just supporting the us qwerty keyboard layout. When inserting a DigiSpark using the DigiKeyboard lib into a computer with qwertz, "y" changes to "z" for example. 

There should also be an option for qwerty and azerty keyboards. I would love to do it myself, but I don't really get the idea of how it is working.

panki27, 2017-01-06:
I'm having the same issue. 
I believe the way to accomplish this is by editing the scancode-ascii-table.h file, so it contains the correct scancodes for a QWERTZ/whatever layout.

For Y/Z, it's as easy as changing every Y in the string you want to enter to a Z and vice versa, however, if you're trying to include characters such as the backslash, it gets very frustrating.

I've tried digging around and finding a correlation between the scancodes in the file and the actual hardware scan codes, but I just don't get it 👎

Fardenco, 2017-02-18:
I had the same issue, I modified a few files to comply with azerty layout.

That's true I also had a hard time finding correlation between scancodes, and I didn't find any.
I've found most of the scancodes for azerty layout on the pictures in this thread
https://digistump.com/board/index.php?topic=2289.0
but "<>" and all altgr keys were missing.

My solution for finding "<>" was simply to write a for loop sending keystrokes for all scancodes from 1 to 255, and looking what each of these were doing until I see "<" on the screen. ">" is simply the same key with shift modifier.

For altgr key combination, I found out that keycodes for keyboard characters fit on the 6 less significant bits of a char. The 1st msb is already used to apply or not shift modifier, I simply used the 2nd msb to apply or not altgr modifier.

But there was a problem, "<" keycode is 100, and this is the only one that does not fit in 6 bits and so it use the bit I just used for altgr modifier. My solution for this is quite dirty but it works : I simply put a condition precisely for "<" and ">" characters case.

Finaly this works, for testing I used the following line :

DigiKeyboard.println("abcdefghijklmnopqrstuvwxyz ABCDEFGHIJKLMNOPQRSTUVWXYZ 1234567890 !\"#$%&'()*+,-./:;<=>?@[\\]^_`{|}~");

and all characters works fine.

Find the files for azerty layout  attached. (added .txt to comply with file post requirements)

[DigiKeyboard.h.txt](https://github.com/digistump/DigistumpArduino/files/784649/DigiKeyboard.h.txt)
[scancode-ascii-table.h.txt](https://github.com/digistump/DigistumpArduino/files/784650/scancode-ascii-table.h.txt)

I suppose you could use the same process for any keyboard layout.

mame82, 2017-02-20:
@Fardenco You're right on the codes of the ASCii relevant key, which are all encoded with values less than 0x65. Thus the second MSB could be used. I haven't done this for a simple reason. The USB descriptor used by DigiKeyboard.h is restricted to send values up to dec 101 (0x65). I changed the descriptor to allow sending values up to dec 231. This allows sending keycodes grater than 0x65 (most of them don't belong to PC AT keyboard, see USB HID keyboard usage table for reference). In the end this allowed me for example to send key 0x80, which on my Linux targets results in a Volume Up keypress and works out of the box with an USB descriptor fix. See [here](lhttps://github.com/hak5darren/USB-Rubber-Ducky/issues/48#issuecomment-279345261) for details.

Fardenco, 2017-02-20:
Oh, that's why I did not managed get the media keys working !
That's great, I'll definitly try this.
But my use of the second msb is just done on the print function, so it should not interfer with your fix !

DerSibi, 2017-08-06:
Hi everyone,

I used Fardenco's excellent previous work and started on a mapping for a qwertz-Layout. It's still not perfect, but most keys work.

Known to not work yet: §|^´`äöüÄÖÜ

As of right now, this is perfectly fine for me so in case anyone still looking for a mapping, I wanted to share. Feel free to work out the remaining issues.

[DigiKeyboard.h.txt](https://github.com/digistump/DigistumpArduino/files/1203239/DigiKeyboard.h.txt)
[scancode-ascii-table.h.txt](https://github.com/digistump/DigistumpArduino/files/1203238/scancode-ascii-table.h.txt)

Have fun!

Sibi

Manurban, 2017-09-07:
Hello,

I am new to arduino. Could you help me to use thoses libraries? How to change the orginal one? I tried to replace the files on my hard drive but i could not find them. Do you know where they are?

Thank U.

Fardenco, 2017-09-08:
You'll find these files in the DigisparkKeyboard library folder.
It should be at "C:\Program Files (x86)\Arduino\libraries\DigisparkKeyboard"
or may be "C:\Users\XXX\Documents\Arduino\libraries\DigisparkKeyboard"

BesoBerlin, 2017-11-28:
hi and thank you first for your work!
so i tried to complete Layout for German Keyboards, but §|`äöüÄÖÜ wont work thi way, as they are not ascii characters. 
^ and ` are working, but i just cant get the |
strange is, identifier 44 in my case works for < and for space as well. when I add MSB it prints > as suspected, but when I add the 2ndMSB for AltGr it doesnt work...
Did one of you guys get it working?

panki27, 2018-04-23:
For anyone still having trouble with this: use duck2spark.
https://github.com/mame82/duck2spark

Fardenco, 2018-04-24:
This is great, thank you for sharing

Cs-hack00, 2021-03-05:
I am not being able to use the TAB key. I have tries KEY_TAB after giving define KEY_TAB
I have used 43,10.Moreover I have also used "/t". Can anyone help plz?

ArminJo, 2021-03-09:
Use https://github.com/ArminJo/DigistumpArduino, this has an extended keyboard example with [international layout support](https://github.com/ArminJo/DigistumpArduino/blob/982824d0f638a0c567a125ae43d4d5632f65e76f/digistump-avr/libraries/DigisparkKeyboard/keylayouts.h#L41).

@likeaguest Can you please close the issue? Thanks

closed 2021-03-09 by likeaguest