digistump/DigistumpArduino #57: runtime.tools.avr-gcc.path and archive_file_path

alcir, 2017-02-11:
Arduino: 1.6.4 (Linux), Board: "Digispark (Default - 16.5mhz)"

> Cannot run program "{runtime.tools.avr-gcc.path}/bin/avr-g++" (in directory "."): error=2, No such file or directory

In `.arduino15/packages/digistump/hardware/avr/1.6.7/platform.txt` I've changed `compiler.path` to `/usr/bin/`, but now I get 

> /usr/bin/avr-ar: {archive_file_path}: Permission denied Error compiling.

Where `archive_file_path` is defined?

MiniLight, 2017-04-27:
I encountered the same problem. 

as a temporary solution, open:
`/home/ml/.arduino15/packages/digistump/hardware/avr/1.6.7`

and edit it:
```
compiler.path=/usr/bin/
archive_file_path=/home/[your username]/[some path]/archive_file
```

n2vram, 2017-06-26:
Given https://forum.arduino.cc/index.php?topic=358313.0
Perhaps a better solution is to add a preference.txt entry with literally:

**archive_file_path={build.path}/{archive_file}**