digistump/DigistumpArduino #60: Cant add digistump boards to any ide not working any more

dunk8888, 2017-03-27:
My ide is missing programmer parameters,any help would be great.does it work with 1.8.1 or 1.6.5?

pfeerick, 2017-04-08:
Programmer parameters? You shouldn't need ot set any programmer parameters or upload using a programmer... merely plug in the digispark when prompted by the Arduino IDE, and then let micronucleus do its thing. And yes, the digistump package does works with Arduino IDE 1.8.1 as I am currently running it, and have been for a while now.

dunk8888, 2017-04-08:
When i try to add the board url it doesnt work in 1.8.1 but in the 1.6.5 it does.

dunk8888, 2017-04-17:
Hi thanks for your reply,its working now all of a sudden not sure what was
wrong,i wasnt being prompted to plug digispark in but its fine now, i
downloaded the 1.6.9 ide and installed on that ide then it seemed to work
ok on 1.8.1 now too.
thanks again.

pfeerick, 2017-04-18:
Might have just been something to do with the board selection in the Arduino IDE... hard to tell without screenshots of what you were seeing when it played up. Anyway, seems like it's working for you again, so that's great to hear.  Can you close the issue since it's no longer an ongoing problem? Thanks ;)

GDI123, 2017-04-18:
Looks like I have the same issue in Arduino IDE 1.8.2. Package manager report "CRC error, file corrupted. Try to repeat later"

pfeerick, 2017-04-19:
I just downloaded the Digispark AVR and SAM packages again on Arduino 1.8.1 (Win10) without any issue. Package manager errors are more often an issue with your download. And unless the Arduino folks have finally fixed the bug, it stays stuck that way unless you download the packages. I'll have to boot into linux to tell you the proper path, but you wanted to navigate to something like ~/.arduino15/staging/packages and just delete the contents of that folder... that is where the files are downloaded and then unpacked

GDI123, 2017-04-19:
Had the same issue. Package manager reported CRC error for files. I've managed to fix issue, because I've already did installation for other user on my PC. So the fix was to copy 'staging/packages' folder to current user. And the issue was in little size of 'avr-gcc-4.8.1....tar.bz2' file.

pfeerick, 2017-04-20:
Drats... so they probably haven't fixed that bug. I thought they said it was fixed a few versions ago... where if a file doesn't pass the CRC check (indicating that the CRC checksum value in the board package hasn't been updated, or that the file hasn't properly downloaded or been tampered with), that it would be deleted/re-downloaded. 

Regardless of that... the avr-gcc dependency is pulled in directly from the Arduino board stuff (as shown [here](https://github.com/digistump/arduino-boards-index/blob/master/package_digistump_index.json#L50)), so it failing to pass the CRC check, download, or install is a Arduino problem... not anything that has been done by Erik/Digistump.