digistump/DigistumpArduino #63: DigiMouse.h: leftClick(), rightClick() and middleClick() all do the same thing

arpruss, 2017-06-20:
In DigistumpArduino/digistump-avr/libraries/DigisparkMouse/DigiMouse.h the definitions of leftClick(), rightClick() and middleClick() are all the same!

mariliahoshino, 2020-07-10:
Yes, It's wrong
the correct is in line 254 until 264, need change these lines

	void rightClick(){
		last_built_report[0] = MOUSEBTN_RIGHT_MASK;
	}

	void leftClick(){
		last_built_report[0] = MOUSEBTN_RIGHT_MASK;
	}
	
	void middleClick(){
		last_built_report[0] = MOUSEBTN_RIGHT_MASK;

mariliahoshino, 2020-07-11:
Finished my project
Mouse USB wireless via App

https://github.com/mariliahoshino/Mouse_USB

https://www.instagram.com/p/CCg9Zs8J6cR/

pfeerick, 2020-07-12:
Looks great! Nice to see new projects showing the Digispark still has a place, even with 'bigger/better' devboards like the ESP8266/NodeMCU! :)