digistump/DigistumpArduino #71: Digikeyboard GUI/Start Button

DarrenRainey, 2017-11-28:
Hello I can't seem to get the digispark with digikeyboard library to press the start button to bring up the menu but it works if I used MOD_GUI_LEFT + KEY R.

I've tried used the keyboards on page 59 of the USB HID manual here  : http://www.usb.org/developers/hidpage/Hut1_12v2.pdf and entering various codes such as 128,231,227

any help or suggestions would be great as I would like my digispark to open the start menu when plugged in

DarrenRainey, 2017-11-28:
This is a workaround I have for windows 10 using MOD_GUI_LEFT + Q but I would prefer a solution that doesn't require the use of another letter such as Q for compatibility

```
#include "DigiKeyboard.h"

void setup() {
DigiKeyboard.sendKeyStroke(0);
DigiKeyboard.sendKeyStroke(KEY_Q, MOD_GUI_LEFT);

DigiKeyboard.println("chrome.exe https://www.youtube.com/embed/QH2-TGUlwu4?rel=0&autoplay=1");
DigiKeyboard.sendKeyStroke(KEY_ENTER);
}

void loop() {
}
```

Phoenix1747, 2018-01-02:
Had this problem earlier today. If you only want the MOD_GUI_LEFT keystroke then you're probably fine with this:
```
DigiKeyboard.sendKeyStroke(0,MOD_GUI_LEFT);
```
This is the best (most clean) solution I found to this _issue_, hope it helps.

DarrenRainey, 2018-01-02:
Yeah I figured that out a while ago. Thanks anyway

closed 2018-01-02 by DarrenRainey