digistump/DigistumpArduino #73: [Requst] - Wiki update

ArcAIN6, 2018-01-02:
I recently recieved a few of these devices, apparently, they are clones, without bootloaders. 

Looking over the wiki, i noticed it is almost completely outdated.  In particular, just about all of the information is based on the now antiquated 1.6x IDE. 

While it might be possible for people to wrangle the particulars for using the new IDE, and burning the bootloader, and fuses on a bare attiny85, etc. from elsewhere. It seems that people like me, who were gifted unfinished, or dodgy parts, there doesn't seem to be much recourse for "fixing" it. 

In my particular case, i received those rev3 clones, and one of them works, but the other 4 do not.
The working board appears to have the 1.6 (assuming 1.06) bootloader on board, however, it won't accept any sketches. The output of the IDE console states the sketch was uploaded successfully with the pretty "thank you" message at the end, but it didn't. I changed the standard "start" file so the blink would be 1000ms on, 50ms off, clicked upload, plugged in the digistump when prompted, uploaded the sketch, the console says it was successful, however, the blink that was already on board does not change.

The non-working 4 appear to not have been programmed at all, and appear to have the stock fuses set for 8mhz external crystal, so i would need to pull the IC off and hit it with a high-voltage programmer to reset the fuses.
At this point, we arrive at the wiki once again, where little to no up-to-date information is available for actually doing this. Everything assumes you have a working digistump. everything assumes you have bootloader onboard already.  There's litteraly nothing about burning the bootloader to a factory fresh attiny85.

It's for these reasons, and i'm sure many others i haven't encountered yet, i think it would be nice for the wiki to be updated.

Thank you.

pfeerick, 2018-01-17:
Not sure if you've posted on the forum, as there is similar comments there as to what appears to be clones marked rev3 which aren't properly programmed there. If micronucleus is running fine other than not accepting new uploads, check the fuse settings, and see if they are correct. If they're not you should be able to follow the [Creating Digispark Compatible clones](https://digistump.com/wiki/digispark/tutorials/programming#creating_digispark_compatible_clones) step - which consists of a Tools -> Burn Bootloader (since attiny85s don't actually have a protected bootloader section of memory, what it really does is set the fuses). Be warned, it will disable the reset pin, hence will need a high voltage programmer afterwards. You can do the high voltage reset without desoldering the attiny85 from the digispark, I have done it before to reset recalcitrant digisparks before.

DeuxVis, 2018-01-18:
Also don't forget that the wiki is a community thing. If you find it lacking something, you can improve it.