digistump/DigistumpArduino #79: Upload failed Error opening

NathanHennigh, 2018-06-11:
I was trying to setup my digispark with arduino ide but when I tried to upload it I got an Error opening.
Below is the full console output

Sketch uses 2810 bytes (46%) of program storage space. Maximum is 6012 bytes.
Global variables use 106 bytes of dynamic memory.
Running Digispark Uploader...
Plug in device now... (will timeout in 60 seconds)
> Please plug in the device ... 
> Press CTRL+C to terminate the program.
> Device is found!
connecting: 16% complete
connecting: 22% complete
connecting: 28% complete
connecting: 33% complete
> Device has firmware version 1.11
> Available space for user applications: 6330 bytes
> Suggested sleep time between sending pages: 8ms
> Whole page count: 99  page size: 64
> Erase function sleep duration: 792ms
> Error opening C:\Users\Nath: No such file or directory
> Error loading or parsing hex file.

pfeerick, 2018-06-12:
Does the `C:\Users\Nath` folder exist? Is there any weird non-standard letters or spaces in the path? You You can turn on verbose messages in the Arduino IDE preferences to see the full path to the hex file (in amongst all the other spew of messages you'll then get) and then maybe get some idea as to what is playing up.

NathanHennigh, 2018-06-13:
No that folder doesnt exist, its supposed to be C:\Users\Nathan Hennigh\...
I never set any path which is weird since it naturally chose one that is wrong.

pfeerick, 2018-06-14:
Ok, I _think_ I see the problem. In the `platform.txt`  file which outlines all the compile and upload command lines, it looks like the path for hex file that gets sent to the digispark isn't as robust as it should be... it looks like it would fail in a situation exactly like yours... where there is a space in the path... as the path hasn't been enclosed in double quotes. 

Try opening `C:\Users\Nathan Hennigh\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\platform.txt` in Notepad++, WordPad, or some other editor that understands linux LF line endings (Notepad will just show it as a wall of text and mangle the file), go down to line 75 `tools.micronucleus.upload.pattern` and see what happens when you change it to 

```tools.micronucleus.upload.pattern="{cmd.path}" -cdigispark --timeout 60 -Uflash:w:"{build.path}/{build.project_name}.hex":i```

I don't have a digispark handy so can't test if this is the proper solution or will fail. Will be two or three days until I can follow this up.

HerrZatacke, 2019-05-09:
This has been over a year. I'm experiencing the exact same issue.
(space in user folder - error message cut's of two letters before the space)

Is there any update?

BidlerShen, 2019-08-23:
**Hacky Fix:**
1. Open Arduino App
2. File -> Preferences
3. Click the preferences.txt and close Arduino App
4. Open preferences.txt
5. Add "build.path=C:\ArduinoOut\" to the file and save
6. Create C:\ArduinoOut folder
7. Relaunch Arduino App
8. Profit

pfeerick, 2019-08-23:
> It is also nice as you can quickly access the Hex Files instead of delving into %APPDATA%

Very true... `Sketch -> Export compiled Binary`  followed by `Sketch -> Show Sketch Folder` works well enough if you want the Hex file, but does mean the sketch needs to be compiled _again_. I suspect changing the `build.path` would fix more than a few errors, including the odd path length error.

HerrZatacke, 2020-02-25:
Any updates on this?

Buleste, 2021-06-23:
This still seems to be an issue.

Sketch uses 718 bytes (11%) of program storage space. Maximum is 6012 bytes.
Global variables use 9 bytes of dynamic memory.
Running Digispark Uploader...
Plug in device now... (will timeout in 60 seconds)
> Please plug in the device ... 
> Press CTRL+C to terminate the program.
> Device is found!
connecting: 16% complete
connecting: 22% complete
connecting: 28% complete
connecting: 33% complete
> Device has firmware version 2.2
> Device signature: 0x1e930b 
> Available space for user applications: 6522 bytes
> Suggested sleep time between sending pages: 7ms
> Whole page count: 102  page size: 64
> Erase function sleep duration: 714ms
> Error opening C:\Users\Dunc: No such file or directory
> Error loading or parsing hex file.

The User should be C:\Users\Duncan Edwards

Any news on a fix for this?

ArminJo, 2021-06-23:
Try https://github.com/ArminJo/DigistumpArduino

Buleste, 2021-06-23:
> 
> 
> Try https://github.com/ArminJo/DigistumpArduino

Same results.

sumukhchitloor, 2022-12-05:
> > Try https://github.com/ArminJo/DigistumpArduino
> 
> Similar results.
> 
> 'C:\Users\Duncan' is not recognized as an internal or external command, operable program or batch file. exit status 1 Error compiling for board Digispark.
> 
> Managed to get the bootloader uploaded by downloading the micronucleus2.5 tools but still no luck because of the space in my Username
> 
> 'C:\Users\Duncan' is not recognized as an internal or external command, operable program or batch file. exit status 1 Error compiling for board Digispark.

Any update on this ?

mszczuro, 2023-03-03:
Fresh installation of the whole IDE, problem persists. I can't work around it as well.
Creating another user just for IDE is no solution for me.
And it's been nearly 5 years now...

viralgupta, 2023-03-04:
After facing the same problem for a year and trying out different solution which didn't work, **I finally found the error and its Fix:**

The problem is with digispark uploader as it doesn't uses 8.3 convention therefore it considers separation characters when parsing text lines, so if you have a space in your username i.e. in your path it will treat it as new line which is causing the issue.

**FIX:**

- Make a folder where all your digispark programs will be saved, this is essential as we have to hardcode the folder location in next steps. I have it in `D:\Digispark\` 

- Now go to Arduino IDE and make a new sketch and delete all the default code as we wont need it and add comment `//Digispark main file` and save the file into the folder where we were going to create digispark files in, For me it was in `D:\Digispark\Digispark.ino` ***make sure the the file name is exactly same as folder name as it is case sensitive** 

- Now locate to `C:\Users\*user name*\AppData\Local\Arduino15\packages\digistump\hardware\avr\`   **(Replace \*user name\* with your username of Windows in which arduino is installed)** into you file manager.
Then further locate to `*x.y.z*\platform.txt` **(Where \*x.y.z\* should be your current version)** and edit the file with any text editor.

- go to line **56** and replace the `{build.path}/{build.project_name}.hex` **with** 
`*digispark folder path*/{build.project_name}.hex` where **_\*digispark folder path\*_ has to be replaced with path of folder** which we created 2 steps ago, for me it is `"D:/Digispark/{build.project_name}.hex"`

- now go to line **74** and replace `-Uflash:w:{build.path}/{build.project_name}.hex:i` **with** `-Uflash:w:*digispark folder path*/{build.project_name}.hex:i` where **_\*digispark folder path\*_ has to be replaced with path of folder** just like previous step, for me it is like this `-Uflash:w:D:/Digispark/{build.project_name}.hex:i`

- Now you can open the Digispark.ino file  with aArduino IDE and click on 3 dots `. . .` on the extreme right of the file name and click **New Tab** and name the tab whatever your file name was and then you can code into that tab whatever you want and upload the code to digispark boards.

- If you want to create new program for digispark board, just go to Digispark folder->Digispark.ino->...->New Tab and there you go!!

***Just make sure you are working on the same sketch(Digispark.ino) and opening new tab on same sketch instead of creating new sketch for each program**

MuhammadUsman15, 2023-05-21:
did any one find a proper solution?

nawazawanswl, 2023-05-28:
**Go to**  C:\Users\**"your user name"**\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\

change the platform.txt as below or copy all the text below to platform.txt ( delete all text from platform.txt then past )

-------------------------------------------------------------------------
# Arduino AVR Core and platform.
# ------------------------------

# For more info:
# https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5---3rd-party-Hardware-specification

name=Digistump AVR Boards
version=1.5.4

# AVR compile variables
# --------------------- 

# Default "compiler.path" is correct, change only if you want to overidde the initial value
compiler.path={runtime.tools.avr-gcc.path}/bin/
compiler.c.cmd=avr-gcc
compiler.c.flags=-c -g -Os -w -ffunction-sections -fdata-sections -MMD
compiler.c.elf.flags=-Os -Wl,--gc-sections
compiler.c.elf.cmd=avr-gcc
compiler.S.flags=-c -g -x assembler-with-cpp
compiler.cpp.cmd=avr-g++
compiler.cpp.flags=-c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -MMD
compiler.ar.cmd=avr-ar
compiler.ar.flags=rcs
compiler.objcopy.cmd=avr-objcopy
compiler.objcopy.eep.flags=-O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0
compiler.elf2hex.flags=-O ihex -R .eeprom
compiler.elf2hex.cmd=avr-objcopy
compiler.ldflags=
compiler.size.cmd=avr-size
# this can be overriden in boards.txt
build.extra_flags=

# AVR compile patterns
# --------------------

## Compile c files
recipe.c.o.pattern="{compiler.path}{compiler.c.cmd}" {compiler.c.flags} -mmcu={build.mcu} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {build.extra_flags} {includes} "{source_file}" -o "{object_file}"

## Compile c++ files
recipe.cpp.o.pattern="{compiler.path}{compiler.cpp.cmd}" {compiler.cpp.flags} -mmcu={build.mcu} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {build.extra_flags} {includes} "{source_file}" -o "{object_file}"

## Compile S files
recipe.S.o.pattern="{compiler.path}{compiler.c.cmd}" {compiler.S.flags} -mmcu={build.mcu} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {build.extra_flags} {includes} "{source_file}" -o "{object_file}"

## Create archives
recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{archive_file_path}" "{object_file}"

## Combine gc-sections, archives, and objects
recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mmcu={build.mcu} -o "C:\ArduinoOut/{build.project_name}.elf" {object_files} "{archive_file_path}" "-L C:\ArduinoOut" -lm

## Create eeprom
recipe.objcopy.eep.pattern="{compiler.path}{compiler.objcopy.cmd}" {compiler.objcopy.eep.flags} "C:\ArduinoOut/{build.project_name}.elf" "C:\ArduinoOut/{build.project_name}.eep"

## Create hex
recipe.objcopy.hex.pattern="{compiler.path}{compiler.elf2hex.cmd}" {compiler.elf2hex.flags} "C:\ArduinoOut/{build.project_name}.elf" "C:\ArduinoOut/{build.project_name}.hex"

## Compute size
recipe.size.pattern="{compiler.path}{compiler.size.cmd}" -A "C:\ArduinoOut/{build.project_name}.elf"
recipe.size.regex=^(?:\.text|\.data|\.bootloader)\s+([0-9]+).*
recipe.size.regex.data=^(?:\.data|\.bss|\.noinit)\s+([0-9]+).*
recipe.size.regex.eeprom=^(?:\.eeprom)\s+([0-9]+).*

# AVR Uploader/Programmers tools
# ------------------------------

#tools.micronucleus.cmd.path={runtime.ide.path}/hardware/digistump/avr/tools/avrdude
#tools.micronucleus.cmd.path.linux={runtime.ide.path}/hardware/digistump/avr/tools/avrdude
tools.micronucleus.cmd.path={runtime.tools.micronucleus.path}/launcher

tools.micronucleus.upload.params.verbose=
tools.micronucleus.upload.params.quiet=
#tools.micronucleus.upload.pattern="{cmd.path}" --run --timeout 60 C:\ArduinoOut\{build.project_name}.hex"
tools.micronucleus.upload.pattern="{cmd.path}" -cdigispark --timeout 60 -Uflash:w:C:\ArduinoOut\{build.project_name}.hex:i
 
# USB Default Flags
# Default blank usb manufacturer will be filled it at compile time
# - from numeric vendor ID, set to Unknown otherwise
build.usb_manufacturer=
build.usb_flags=
------------------------------------------------------------------------------------------
copy text only not lines.......................it is working perfectly with Arduino IDE 2.10
hot it helps

nawazawanswl, 2023-05-28:
platform.txt

`# Arduino AVR Core and platform.
# ------------------------------

# For more info:
# https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5---3rd-party-Hardware-specification

name=Digistump AVR Boards
version=1.5.4

# AVR compile variables
# --------------------- 

# Default "compiler.path" is correct, change only if you want to overidde the initial value
compiler.path={runtime.tools.avr-gcc.path}/bin/
compiler.c.cmd=avr-gcc
compiler.c.flags=-c -g -Os -w -ffunction-sections -fdata-sections -MMD
compiler.c.elf.flags=-Os -Wl,--gc-sections
compiler.c.elf.cmd=avr-gcc
compiler.S.flags=-c -g -x assembler-with-cpp
compiler.cpp.cmd=avr-g++
compiler.cpp.flags=-c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -MMD
compiler.ar.cmd=avr-ar
compiler.ar.flags=rcs
compiler.objcopy.cmd=avr-objcopy
compiler.objcopy.eep.flags=-O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0
compiler.elf2hex.flags=-O ihex -R .eeprom
compiler.elf2hex.cmd=avr-objcopy
compiler.ldflags=
compiler.size.cmd=avr-size
# this can be overriden in boards.txt
build.extra_flags=

# AVR compile patterns
# --------------------

## Compile c files
recipe.c.o.pattern="{compiler.path}{compiler.c.cmd}" {compiler.c.flags} -mmcu={build.mcu} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {build.extra_flags} {includes} "{source_file}" -o "{object_file}"

## Compile c++ files
recipe.cpp.o.pattern="{compiler.path}{compiler.cpp.cmd}" {compiler.cpp.flags} -mmcu={build.mcu} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {build.extra_flags} {includes} "{source_file}" -o "{object_file}"

## Compile S files
recipe.S.o.pattern="{compiler.path}{compiler.c.cmd}" {compiler.S.flags} -mmcu={build.mcu} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {build.extra_flags} {includes} "{source_file}" -o "{object_file}"

## Create archives
recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{archive_file_path}" "{object_file}"

## Combine gc-sections, archives, and objects
recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mmcu={build.mcu} -o "C:\ArduinoOut/{build.project_name}.elf" {object_files} "{archive_file_path}" "-L C:\ArduinoOut" -lm

## Create eeprom
recipe.objcopy.eep.pattern="{compiler.path}{compiler.objcopy.cmd}" {compiler.objcopy.eep.flags} "C:\ArduinoOut/{build.project_name}.elf" "C:\ArduinoOut/{build.project_name}.eep"

## Create hex
recipe.objcopy.hex.pattern="{compiler.path}{compiler.elf2hex.cmd}" {compiler.elf2hex.flags} "C:\ArduinoOut/{build.project_name}.elf" "C:\ArduinoOut/{build.project_name}.hex"

## Compute size
recipe.size.pattern="{compiler.path}{compiler.size.cmd}" -A "C:\ArduinoOut/{build.project_name}.elf"
recipe.size.regex=^(?:\.text|\.data|\.bootloader)\s+([0-9]+).*
recipe.size.regex.data=^(?:\.data|\.bss|\.noinit)\s+([0-9]+).*
recipe.size.regex.eeprom=^(?:\.eeprom)\s+([0-9]+).*

# AVR Uploader/Programmers tools
# ------------------------------

#tools.micronucleus.cmd.path={runtime.ide.path}/hardware/digistump/avr/tools/avrdude
#tools.micronucleus.cmd.path.linux={runtime.ide.path}/hardware/digistump/avr/tools/avrdude
tools.micronucleus.cmd.path={runtime.tools.micronucleus.path}/launcher

tools.micronucleus.upload.params.verbose=
tools.micronucleus.upload.params.quiet=
#tools.micronucleus.upload.pattern="{cmd.path}" --run --timeout 60 C:\ArduinoOut\{build.project_name}.hex"
tools.micronucleus.upload.pattern="{cmd.path}" -cdigispark --timeout 60 -Uflash:w:C:\ArduinoOut\{build.project_name}.hex:i
 
# USB Default Flags
# Default blank usb manufacturer will be filled it at compile time
# - from numeric vendor ID, set to Unknown otherwise
build.usb_manufacturer=
build.usb_flags=
`

nawazawanswl, 2023-05-28:
Attached platform.txt
[platform.txt](https://github.com/digistump/DigistumpArduino/files/11584123/platform.txt)

Yogeshroy, 2023-07-12:
the problem is with the launcher and micronucleus executables included with the digistump package as they fail to parse paths with spaces properly. you can use the latest micronucleus executable from the ATTinyCore package. replace the micronucleus executable with the latest one and make the following change to platform.txt.

line 70: tools.micronucleus.cmd.path={runtime.tools.micronucleus.path}\launcher
change to: tools.micronucleus.cmd.path={runtime.tools.micronucleus.path}\micronucleus

line 75: tools.micronucleus.upload.pattern="{cmd.path}" -cdigispark --timeout 60 -Uflash:w:{build.path}/{build.project_name}.hex:i
change to : tools.micronucleus.upload.pattern="{cmd.path}" --timeout 60 --run --no-ansi "{build.path}/{build.project_name}.hex"

Hope it helps