digistump/DigistumpArduino #80: Adding other keyboard templates

whoismath, 2018-07-08:
Hi, i'm writing an ABNT2 version of this driver (https://github.com/whoismath/DigistumpArduino) and i'm wondering if you guys would like to add others layout keyboards into your projects, my version is not 100% working but it's working and getting better. But seems that you don't accept new pull requests since 2016, are you still active in this project? can i submit abnt2 (portuguese) layout into your project?

rsrdesarrollo, 2018-11-13:
you might check out my fork (https://github.com/rsrdesarrollo/DigistumpArduino) with full keyboard layouts support.

ArminJo, 2020-06-11:
This request is included in the new [version 1.6.8](https://github.com/ArminJo/DigistumpArduino) of Digistump avr core.

Please could you be so kind and close the issue, since this repo is abandoned.
Thank you!
Armin