digistump/DigistumpArduino #84: Question: What "micronucleus" programmer does?

Harvie, 2019-01-15:
What is this for? Can i use micronucleus enabled board as ISP programmer to flash other boards?
Have you been thinking about packaging micronucleus with this Arduino package, so i can flash latest micronucleus directly from arduino IDE using "burn bootloader"?

[image]

bratoff, 2019-01-15:
Micronucleus is not a programmer - it's the bootloader that's provided on the Digispark and Digispark Pro boards.  I'd imagine it was added to the programmer list just so that you could upload to those boards.

BTW, Micronucleus is not proprietary to Digistump and can be found on Github.  It supports a number of the ATtiny chips and includes an embedded software-based USB stack, making it useful for AVR chips that lack the resources to run Optiboot (the current Arduino bootloader).

Harvie, 2019-01-16:
> Micronucleus is not a programmer 

I know. That's why i am asking why it's in the programmer list in Arduino :-)

> I'd imagine it was added to the programmer list just so that you could upload to those boards.

I know, but that's not usualy done through "programmer" menu in Arduino IDE. If you upload sketch using bootloader, you are supposed to use "Sketch -> Upload (CTRL+U)" rather than programmer infrastructure, which is on contrary commonly used to update/fix bootloader using ISP.

> Micronucleus is open source and can be found in its own repository on Github

I know. I use it.

But i think it would make sense to include it to DigiSpark board package. Just like Arduino core includes optiboot (and/or atmegaboot) which is separate project just like micronucleus.

bratoff, 2019-01-16:
I don't work for Digistump, so I can't speak to why they don't include Micronucleus in their board package.  If you look on their wiki, it does have instructions for reflashing the Digisparks, including the download link to the down-rev version of Micronucleus that comes on the boards.  However, Digisparks ship with the Reset Disable fuse set, making one-click reflashing impossible unless you have access to a HVSP programmer or have built one of the HVSP rescue circuits floating around the web.  (Maybe that's why they don't package their bootloader?)
Erik appears to be all tied up on his 3D printer project, but maybe when he comes up for air he can provide additional insight.

pfeerick, 2019-01-16:
It _is_ included in the board package - try looking the `tools` folder. I beleive the reason it is listed in the 'programmer' menu is due to a quirk of how you tell the Arduino IDE how to actually program the board - it isn't using a built-in programming method, so you provide your own upload method via a custom programmer. You'll see the `digispark-tiny.upload.using=micronucleusprog` in `boards.txt` which corresponds with `micronucleusprog.name=Micronucleus` in `programmers.txt`.

bratoff, 2019-01-16:
Makes sense - I think that was the only way to do it in Arduino 1.6.5.  In the latest Arduino releases, you can  add tool formulas to platform.txt and reference them from boards.txt.

Harvie, 2019-01-16:
> It's probably not shipped because the digisparks are all pre-flashed

My use case is that i want to make custom PCB with attiny85 along with some other stuff, so i have everything on single board without need to bareback extra digispark board on it. So it would make sense to be able to flash it. But i guess that i wouldn't like the idea that this disables ISP in the process.

bratoff, 2019-01-16:
If you're making your own board, starting with unprogrammed attiny85s, you don't have to disable ISP - just don't set the Disable Reset fuse after programming.  Actually, if you have the ability to use ISP, why do you need Micronucleus (or any bootloader) at all?

Harvie, 2019-01-17:
> why do you need Micronucleus (or any bootloader) at all?

Why do i use the Arduino IDE in the first place? Why do i need to write programs in C when i can use ASM? Why even i bother to use attiny for something that can be achived with bunch of logic gates? Because i am lazy and it's convenient! :-)

ISP means aditional wiring. I usualy use some kind of makeshift wiring to board that does not have ISP connnections ready. Sometimes i remove attiny from socket and put it into breadboard to flash it. That's annoying, and i already have V-USB wired, so why not use it. No need to fiddle with additional programmer HW. Also you can have smaller PCB when you don't need to route ISP connection.

pfeerick, 2019-01-17:
> Why do i use the Arduino IDE in the first place?

Yes, indeed, what is wrong with you, wanting life to be easy? :-P 

Compile your own version of micronuculeus if you want any custom settings (clock speed, jumper to go into bootloader) or just used the [latest binary](https://github.com/micronucleus/micronucleus/tree/master/firmware/releases) - it's now v2.04 rather than the 1.06-ish it was when the Digispark was initially released. Then use avrdude or a gui frontend like  avrdudess to set your fuses (just don't disable the RST pin), and program your bare attiny85s via ICSP before putting them into your project board. All done, and ready for v-usb / micronucleus if you need it in the future.

bratoff, 2019-01-18:
> > Compile your own version of micronuculeus
> 
> I've already done that LOL. But this is "DigistumpArduino" repository :-)

Yes, and your board is not a Digistump board!  As you correctly point out, this discussion belongs on the Micronucleus repository.

bratoff, 2019-01-18:
@Harvie - I just looked at the Digistump website and they are still selling Digisparks - they're just sold out at the moment. Anyway, I don't see it as arrogance. If the buyer won't support you, why should you support them?

ArminJo, 2020-05-04:
The possibility to upgrade a bootloader is now included in the new [version 1.6.8](https://github.com/ArminJo/DigistumpArduino) of Digistump avr core.

Have fun

Harvie, 2020-05-04:
> If the buyer won't support you, why should you support them?

That is based on the premise that only possible support is the financial one. People caring about your ecosystem, figuring improvements, reccommending it to other people, preparing pull requests for you and improving the thing in other ways are support too. I might not buyed your board, but few months ago i've contributed some time (=money) to few arduino libraries in order to make them buildable for your board, which means your boards will now have more value for buyers and therefore might theoreticaly provide little bit more revenue in the long run (if money is important for you)...