digistump/DigistumpArduino #85: Warning! The USB Examples breaks the bootloader USB enumeration

E3V3A, 2019-01-16:
I just found this repo, after having posted my issue [here](https://github.com/micronucleus/micronucleus/issues/144). 

#### The Summary: 
The example code for using USB, will break the Digispark bootloaer enumeration, and will make your (Windows) machine think that your Digistump is a `DigiUSB`, instead of a bootlaoder, preventing any future connections and (Arduino IDE) uploads. 

* How can this be resolved?  
(I already tried to re-install all the device drivers, obviously to no avail, since the micronucleus is not smart enough to deal with this.)

**Please put some warnings in these example codes, to make that clear to the user, before he flashes over his boot loader connections!**

bratoff, 2019-01-17:
I do not work for Digistump, but please allow me to add my 2 cents worth:  The board pictured in your post at the micronucleus repo is NOT a Digistump product. Is it really fair to expect them to support it?

E3V3A, 2019-01-17:
> The board pictured in your post at the micronucleus repo is NOT a Digistump product. Is it really fair to expect them to support it?

Yes, because it is exactly the same circuit as in the (original) Digistump, except the ugly PCB printed USB connector is replaced by a proper uUSB, thus even a better product. Everything else is the same. (And since I has no clue that this was a knock-off, it seem that Digistump still get all the credit with whatever projects are created by these.)

No, **I no longer expect anyone to support anything!** (Even RPi people doesn't support their own product issues.)

pfeerick, 2019-01-18:
Ah yes, RPi support... lets not go there... it can be quite confusing at times... making you think it would be much more fun watching paint dry... Not to say it can't be good... but when it isn't... it's not 😮  Does that make sense? 😆 

Anyway, lets be frank - nothing is likely to happen here as this repo hasn't been updated in the last two years. But you do have a valid point - Windows USB enumeration is that slow at times that the 5 seconds could be over before the computer even makes a device available for programming so that micronucleus can do it's think... hence a warning note for Windows users should be included... probably indicating that they will need to either become familar with the HVSP process and have a ICP (via an Arduino or actual programmer) to reset the device if it doesn't co-operate, or be able to get a live linux system working to have faster USB enumeration speeds.

E3V3A, 2019-01-23:
Yep, it seem that this repo has been abandoned. No PR merges for years, with 11 outstanding. 
Not sure this place serve an other purpose than a /dev/null for bitching and whining! 
:sob:  

The examples may still be useful, but now orphaned.

phamtienmanh, 2019-02-24:
> sure this place serve an other purpose than a /dev/null for bitching and whining!
> 😭
> 
> The examples may still be useful, but now orphaned.

Hello E33V3A, did u solve the problem
I followed this tutorial https://www.electromaker.io/blog/article/programming-ws2812b-leds-with-attiny85-23 , then I cant have my attiny85 as usb bootloader  anymore. I guess it cause of using pin number 4, with is USB-. Need some help plz, I am not an arduino man

pfeerick, 2019-02-24:
Which of these is your board?

[image]

If it's the ones on the left, then you shouldn't be having the problem unless the a) the board is faulty or b) you're having issues with your computer not detecting the digispark before the bootloader times out. If it's the one on the right, you may be able to reprogram the bootloader as the Chinese are lazy and don't bother setting the proper fuses, but that would require some Arduino knowhow...