digistump/DigistumpArduino #92: Sketch not persistent

Bobofiero, 2019-05-18:
When i uploade a sketch into my attiny85 it work only if the board remain connected to PC: if i put it off and put in the sketch doesn't start.
Why this happened??

bratoff, 2019-11-05:
Are you waiting long enough (about a minute) for the micronucleus bootloader to time out?  Every time you plug the digispark in, the bootloader runs and waits for about a minute for a new upload.  After it times out, your sketch should run.

ArminJo, 2020-06-02:
The bootloader waits for **6 second**s, not longer!
But what is your attiny85 and what bootloader do you use?
Consider to use the new [version 1.6.8](https://github.com/ArminJo/DigistumpArduino) of Digistump avr core.