digistump/DigistumpArduino #95: Error downloading http://digistump.com/package_digistump_index.json

Bora66, 2019-06-22:
I have been getting this error msg in Arduino IDE for some time now: 

Error downloading http://digistump.com/package_digistump_index.json

Digistump boards have disappeared from the board list and and "Digistump AVR Boards" is gone from boards manager/contributed.

I have tried deleting and re-writing the URL line from Files/Preferences/Additional Boards Manager ULSs
I have tried adding the redirected URL https://raw.githubusercontent.com/digistump/arduino-boards-index/master/package_digistump_index.json
Both resulted with the "Error downloading..." msg.

Both URL's open fine in my browser, so I know the json file is there... 

What's going on ??

asjdf, 2019-06-22:
so do i.i think it's arduino's problem

pfeerick, 2019-06-22:
Sorry, can't reproduce (on Linux) - both of the URLs works just fine with Arduino IDE 1.8.9 for me. File->Preferences, add URL, Ok, Tools -> Board, Board Manager, wait for refresh, all three board families show (AVR,SAM,Oak). I'll try again on Windows later.

Maybe you just caught the Arduino IDE or your internet having having a momentary spastic. 

```
http://digistump.com/package_digistump_index.json
https://raw.githubusercontent.com/digistump/arduino-boards-index/master/package_digistump_index.json
```

geebee95, 2019-07-25:
Hi Bora66 - did you ever sort this?  I'm having the same problem trying to use the AT Tiny85 board.  I have tried and tried and tried to get the board to be recognised.  No matter what I try I ALWAYS get the 'Errror downloading...' message.  This happens with the .JSON and the .ZIP files. It's driving me crazy now.  I'm using Win10 and Arduino 1.8.9.  Can anyone help please.  Thanks

gaodazhuangde, 2019-10-27:
I am also experiencing the same problem. Because I don’t know if someone has solved it, share the method. Thank you.

gigawatts, 2019-11-09:
https://github.com/esp8266/Arduino  supports the Oak now, just use that.

pfeerick, 2019-11-09:
I've been using the ESP8266 core since the board profile for the Oak was added and can also second that IF you are using the Oak, However, this package also supports the SAM and AVR boards, so that isn't an option if you use those boards.

ningman, 2020-07-17:
Same problem ,anyone knows?

w4iting4, 2024-03-25:
When I use 1.6.4 IDE I can't download from https://raw.githubusercontent.com/digistump/arduino-boards-index/master/package_digistump_index.json or http://digistump.com/package_digistump_index.json, either url Neither can work。help ，plz 🤕

Nikita-Goncharov, 2024-03-27:
I had that problem too. This site can be useful:
[https://www.linuxndroid.com/2022/12/not-downloading-how-to-fix-digistump.html](https://www.linuxndroid.com/2022/12/not-downloading-how-to-fix-digistump.html)

Note: add .json file ONLY in this directory **C:\Users\YourName\AppData\Local\Arduino15**