digistump/DigistumpArduino #97: Can't compile a project with SoftSerial and DigiKeyboard

prk3, 2019-07-20:
Arduino IDE 1.8.9
Digistump AVR Boards 1.6.7

I've got a sketch with 3 includes:
```c
#include <Arduino.h>
#include <DigiKeyboard.h>
#include <SoftSerial.h>
```

I get this linker error while compiling the sketch:
```
libraries/DigisparkTinyPinChange/TinyPinChange.cpp.o: In function `__vector_2':
/home/user/.arduino15/packages/digistump/hardware/avr/1.6.7/libraries/DigisparkTinyPinChange/TinyPinChange.cpp:57: multiple definition of `__vector_2'
libraries/DigisparkKeyboard/usbdrvasm.S.o:/home/user/.arduino15/packages/digistump/hardware/avr/1.6.7/libraries/DigisparkKeyboard/usbdrvasm165.inc:41: first defined here
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Digispark (Default - 16.5mhz).
```

Maybe that's a missing `#ifndef`? Or an accidental name collision?

prk3, 2019-07-21:
I wrote my own version of SoftSerial and I it still didn't work. The issue disappeared after removing the external interrupt handler. On ATtiny85 there is only one external interrupt and possibly both libraries define handlers for it.

closed 2019-07-21 by prk3

Giova02, 2021-09-14:
I got a similar problem trying using both TinyWire and DigiKeyboard but idk how to solve it

`libraries\TinyWire-master\twi.cpp.o: In function `__vector_2':
C:\Users\10293_000\Documents\Arduino\libraries\TinyWire-master/twi.cpp:822: multiple definition of `__vector_2'
libraries\DigisparkKeyboard\usbdrvasm.S.o:C:\Users\10293_000\Documents\ArduinoData\packages\digistump\hardware\avr\1.6.7\libraries\DigisparkKeyboard/usbdrvasm165.inc:41: first defined here
collect2.exe: error: ld returned 1 exit status
exit status 1
Errore durante la compilazione per la scheda Digispark (Default - 16.5mhz).
`