digistump/DigistumpArduino #101: EEPROM.h: No such file or directory

LightningStalker, 2019-10-07:
I have installed 1.6.7, Arduino IDE 1.8.10, Arduino core 1.8.1

#include <EEPROM.h>

https://github.com/digistump/DigistumpArduino/issues/34#issue-154927992

pfeerick, 2019-10-07:
And I'm guessing an AVR board? Tiny85?

LightningStalker, 2019-10-07:
Yes the Digispark

pfeerick, 2019-10-07:
It was actually fixed for the AVR boards in commit https://github.com/digistump/DigistumpArduino/commit/b9a79d8b2eacd1578e9788403f90257cc40c836e but unfortunately that 'next release' Ghost commented on in #34 hasn't happened. So grabbing the eeprom library as linked in that issue, or pulling it from the repo manually is your best bet for now.

LightningStalker, 2019-10-07:
That did work. Dropped it in libraries directory.

bratoff, 2019-11-05:
Adding it to your libraries folder isn't necessary.  It's not being found because AVR-specific headers are in a separate folder.  Change the #include statement to:

#include <avr/eeprom.h>

and your code should compile.

pfeerick, 2019-11-05:
It's not being found because the Arduino EEPROM library is not present in the 1.6.7 release, and was added in a later commit, with no subsequent release. Including the core `avr\eeprom.h` library isn't going to help much if the user wants the `.put()/.get()` functions provided by the Arduino EEPROM library, etc.

bratoff, 2019-11-05:
For new code it's just as well to use the standard avr-libc eeprom library. Makes your code more portable too.

pfeerick, 2019-11-05:
Quite the opposite... you just made the code AVR specific, rather than it being cross-platform by using the higher level library. It would no longer work on ESP8266, ESP32, Teensy and other platforms.

bratoff, 2019-11-05:
I meant portability within the AVR world. Non-AVR processors introduce many other issues besides this one.
Anyway, with Erik off the map indefinitely, sticking to avr-libc libraries sure looks like a safer bet to me.
Unless of course someone wants to fork the project and become the new leading edge. You up for that?

pfeerick, 2019-11-05:
True... but I prefer to write platform agnostic as much as I can now, so that the code is as reusable as possible. 

The other option is to simply pull the current version of this core directly from github, rather than use the release, because the library is present there. 

lol... I've been doing stuff on and off for a while now, and have another version of this repo with most of the issues mentioned theoretically fixed, but just haven't find time to test it all and push a release.

ArminJo, 2020-05-02:
Hi Peter, Hi Bruce,
I put all improvements in my [fork](https://github.com/ArminJo/DigistumpArduino) maybe you can test it?
Thanks
Armin