digistump/DigistumpArduino #104: Can't use it with arduino-cli

XhstormR, 2019-11-16:
The following error occurs when using arduino-cli:
```
C:\Users\Leo\test>arduino-cli compile --fqbn digistump:avr:digispark-tiny sketch_nov14a
Sketch uses 2758 bytes (45%) of program storage space. Maximum is 6012 bytes.
Global variables use 96 bytes of dynamic memory.
Error during build: copying output file: open C:\Users\Leo\AppData\Local\Temp\arduino-sketch-900304DE8A4863510F9A5FC3C2DE273D/{recipe.output.tmp_file}: The system cannot find the file specified.

C:\Users\Leo\test>arduino-cli upload --fqbn digistump:avr:digispark-tiny sketch_nov14a -p COM1
Error during Upload: property 'recipe.output.tmp_file' not defined
```
I found that this bug has been fixed,  see https://github.com/arduino/arduino-cli/issues/193, but it has not been released for a long time. Can you release a new version?

ryarnyah, 2020-01-29:
You need to add `recipe.output.tmp_file={build.project_name}.hex` inside `packages/digistump/hardware/avr/1.6.7/platform.txt`

ArminJo, 2020-05-02:
Try the [optimized digistump version](https://github.com/ArminJo/DigistumpArduino), it fixes the error and enables around 20% more FLASH.

ArminJo, 2020-06-11:
Please could you be so kind and close the issue, since this repo is abandoned.
Thank you!
Armin

closed 2020-06-12 by XhstormR