digistump/DigistumpArduino #105: bad CPU type in executable

siliconchris1973, 2019-12-12:
Hello All,

I installed the digispark board as per digistump wiki and wanted to try the digiblink example, when I got above error.

below on may find my full error text.

Can anyone help???

`Arduino: 1.8.10 (Mac OS X), Board: "Digispark (Default - 16.5mhz)"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/chris/Library/Arduino15/packages -hardware /Users/chris/Developer/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/chris/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/chris/Developer/Arduino/libraries -fqbn=digistump:avr:digispark-tiny -ide-version=10810 -build-path /var/folders/sf/g25yg3ys00q875038hypylbr0000gn/T/arduino_build_33939 -warnings=all -build-cache /var/folders/sf/g25yg3ys00q875038hypylbr0000gn/T/arduino_cache_642655 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.micronucleus.path=/Users/chris/Library/Arduino15/packages/digistump/tools/micronucleus/2.0a4 -prefs=runtime.tools.micronucleus-2.0a4.path=/Users/chris/Library/Arduino15/packages/digistump/tools/micronucleus/2.0a4 -prefs=runtime.tools.avr-gcc.path=/Users/chris/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5 -prefs=runtime.tools.avr-gcc-4.8.1-arduino5.path=/Users/chris/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5 -verbose /Users/chris/Library/Arduino15/packages/digistump/hardware/avr/1.6.7/libraries/DigisparkUSB/examples/DigiBlink/DigiBlink.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/chris/Library/Arduino15/packages -hardware /Users/chris/Developer/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/chris/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/chris/Developer/Arduino/libraries -fqbn=digistump:avr:digispark-tiny -ide-version=10810 -build-path /var/folders/sf/g25yg3ys00q875038hypylbr0000gn/T/arduino_build_33939 -warnings=all -build-cache /var/folders/sf/g25yg3ys00q875038hypylbr0000gn/T/arduino_cache_642655 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.micronucleus.path=/Users/chris/Library/Arduino15/packages/digistump/tools/micronucleus/2.0a4 -prefs=runtime.tools.micronucleus-2.0a4.path=/Users/chris/Library/Arduino15/packages/digistump/tools/micronucleus/2.0a4 -prefs=runtime.tools.avr-gcc.path=/Users/chris/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5 -prefs=runtime.tools.avr-gcc-4.8.1-arduino5.path=/Users/chris/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5 -verbose /Users/chris/Library/Arduino15/packages/digistump/hardware/avr/1.6.7/libraries/DigisparkUSB/examples/DigiBlink/DigiBlink.ino
Using board 'digispark-tiny' from platform in folder: /Users/chris/Library/Arduino15/packages/digistump/hardware/avr/1.6.7
Using core 'tiny' from platform in folder: /Users/chris/Library/Arduino15/packages/digistump/hardware/avr/1.6.7
Detecting libraries used...
/Users/chris/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=16500000L -DARDUINO=10810 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR -I/Users/chris/Library/Arduino15/packages/digistump/hardware/avr/1.6.7/cores/tiny -I/Users/chris/Library/Arduino15/packages/digistump/hardware/avr/1.6.7/variants/digispark /var/folders/sf/g25yg3ys00q875038hypylbr0000gn/T/arduino_build_33939/sketch/DigiBlink.ino.cpp -o /dev/null
fork/exec /Users/chris/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-g++: bad CPU type in executable
Fehler beim Kompilieren für das Board Digispark (Default - 16.5mhz).`

kfigiela, 2019-12-20:
Note that macOS Catalina dropped support for 32 bit binaries. Toolchain has to be upgraded to 64 bit architecture.

eziologico, 2020-03-27:
When will Toolchain upgraded to 64 bit architecture to make it working with MacOS Catalina 10.15?
It would be nice!

ArminJo, 2020-05-02:
Can you try the [optimized digistump version](https://github.com/ArminJo/DigistumpArduino) and report the errors here?

eziologico, 2020-05-04:
Hi @ArminJo 
Thank you for your reply and your job.
I upgraded to the new version 1.6.8 of Digistump core for Arduino following your instructions on my iMac with macOS Catalina 10.15.4 and Arduino IDE 1.8.12
So, just compiling any sketch I get the following error:
```
Arduino:1.8.12` (Mac OS X), Scheda:"Digispark (Default - 16.5mhz)"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/ezio/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/ezio/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/ezio/Documents/Arduino/libraries -fqbn=digistump:avr:digispark-tiny -ide-version=10812 -build-path /var/folders/c4/z92mlq012sb_p89mpfdx5k900000gn/T/arduino_build_475889 -warnings=none -build-cache /var/folders/c4/z92mlq012sb_p89mpfdx5k900000gn/T/arduino_cache_122291 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.micronucleus.path=/Users/ezio/Library/Arduino15/packages/digistump/tools/micronucleus/2.0a4 -prefs=runtime.tools.micronucleus-2.0a4.path=/Users/ezio/Library/Arduino15/packages/digistump/tools/micronucleus/2.0a4 -prefs=runtime.tools.avr-gcc.path=/Users/ezio/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/Users/ezio/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -verbose /Users/ezio/Library/Arduino15/packages/digistump/hardware/avr/1.6.8/libraries/Digispark_Examples/Start/Start.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/ezio/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/ezio/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/ezio/Documents/Arduino/libraries -fqbn=digistump:avr:digispark-tiny -ide-version=10812 -build-path /var/folders/c4/z92mlq012sb_p89mpfdx5k900000gn/T/arduino_build_475889 -warnings=none -build-cache /var/folders/c4/z92mlq012sb_p89mpfdx5k900000gn/T/arduino_cache_122291 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.micronucleus.path=/Users/ezio/Library/Arduino15/packages/digistump/tools/micronucleus/2.0a4 -prefs=runtime.tools.micronucleus-2.0a4.path=/Users/ezio/Library/Arduino15/packages/digistump/tools/micronucleus/2.0a4 -prefs=runtime.tools.avr-gcc.path=/Users/ezio/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/Users/ezio/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -verbose /Users/ezio/Library/Arduino15/packages/digistump/hardware/avr/1.6.8/libraries/Digispark_Examples/Start/Start.ino
Using board 'digispark-tiny' from platform in folder: /Users/ezio/Library/Arduino15/packages/digistump/hardware/avr/1.6.8
Using core 'tiny' from platform in folder: /Users/ezio/Library/Arduino15/packages/digistump/hardware/avr/1.6.8
Detecting libraries used...
/Users/ezio/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -flto -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=16500000L -DARDUINO=10812 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR -I/Users/ezio/Library/Arduino15/packages/digistump/hardware/avr/1.6.8/cores/tiny -I/Users/ezio/Library/Arduino15/packages/digistump/hardware/avr/1.6.8/variants/digispark /var/folders/c4/z92mlq012sb_p89mpfdx5k900000gn/T/arduino_build_475889/sketch/Start.ino.cpp -o /dev/null
Generating function prototypes...
/Users/ezio/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -flto -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=16500000L -DARDUINO=10812 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR -I/Users/ezio/Library/Arduino15/packages/digistump/hardware/avr/1.6.8/cores/tiny -I/Users/ezio/Library/Arduino15/packages/digistump/hardware/avr/1.6.8/variants/digispark /var/folders/c4/z92mlq012sb_p89mpfdx5k900000gn/T/arduino_build_475889/sketch/Start.ino.cpp -o /var/folders/c4/z92mlq012sb_p89mpfdx5k900000gn/T/arduino_build_475889/preproc/ctags_target_for_gcc_minus_e.cpp
/Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /var/folders/c4/z92mlq012sb_p89mpfdx5k900000gn/T/arduino_build_475889/preproc/ctags_target_for_gcc_minus_e.cpp
Sto compilando lo sketch...
/Users/ezio/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -flto -MMD -mmcu=attiny85 -DF_CPU=16500000L -DARDUINO=10812 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR -I/Users/ezio/Library/Arduino15/packages/digistump/hardware/avr/1.6.8/cores/tiny -I/Users/ezio/Library/Arduino15/packages/digistump/hardware/avr/1.6.8/variants/digispark /var/folders/c4/z92mlq012sb_p89mpfdx5k900000gn/T/arduino_build_475889/sketch/Start.ino.cpp -o /var/folders/c4/z92mlq012sb_p89mpfdx5k900000gn/T/arduino_build_475889/sketc [... cut ...]
  digitalWrite(0, HIGH);   // turn the LED on (HIGH is the voltage level)
  digitalWrite(1, HIGH);
  delay(1000);               // wait for a second
  digitalWrite(0, LOW);    // turn the LED off by making the voltage LOW
  digitalWrite(1, LOW); 
  delay(1000);               // wait for a second
}
```

But it doesn't matter whatever sketch I compile, even with an empty one I get the same error.
```
void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

Thank you for your help,

Ezio

eziologico, 2020-05-04:
Hi @ArminJo 
Upgrading the new version 1.6.8 of Digistump core for Arduino on my Windows 10 (64bit) laptop with Arduino IDE 1.18.12 it still works well without any issue and I upgraded my ATTiny85 boards Bootloader to version 2.5 also.

Thank you,

Ezio

ArminJo, 2020-05-04:
Hi Ezio
thank you very much for your valuable feedback.
Yes I made an error and forgot to delete one " in platform.txt
I just fixed it

Thanks
Armin

eziologico, 2020-05-04:
Hi @ArminJo 
Yes, it works!!!
I fixed my platform.txt file deleting the " in excess on line 84, so it compiles now and finally it uploads the program to the board.

```
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/ezio/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/ezio/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/ezio/Documents/Arduino/libraries -fqbn=digistump:avr:digispark-tiny -ide-version=10812 -build-path /var/folders/c4/z92mlq012sb_p89mpfdx5k900000gn/T/arduino_build_771399 -warnings=none -build-cache /var/folders/c4/z92mlq012sb_p89mpfdx5k900000gn/T/arduino_cache_890818 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.micronucleus.path=/Users/ezio/Library/Arduino15/packages/digistump/tools/micronucleus/2.0a4 -prefs=runtime.tools.micronucleus-2.0a4.path=/Users/ezio/Library/Arduino15/packages/digistump/tools/micronucleus/2.0a4 -prefs=runtime.tools.avr-gcc.path=/Users/ezio/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/Users/ezio/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -verbose /Users/ezio/Library/Arduino15/packages/digistump/hardware/avr/1.6.8/libraries/Digispark_Examples/Start/Start.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/ezio/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/ezio/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/ezio/Documents/Arduino/libraries -fqbn=digistump:avr:digispark-tiny -ide-version=10812 -build-path /var/folders/c4/z92mlq012sb_p89mpfdx5k900000gn/T/arduino_build_771399 -warnings=none -build-cache /var/folders/c4/z92mlq012sb_p89mpfdx5k900000gn/T/arduino_cache_890818 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.micronucleus.path=/Users/ezio/Library/Arduino15/packages/digistump/tools/micronucleus/2.0a4 -prefs=runtime.tools.micronucleus-2.0a4.path=/Users/ezio/Library/Arduino15/packages/digistump/tools/micronucleus/2.0a4 -prefs=runtime.tools.avr-gcc.path=/Users/ezio/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/Users/ezio/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -verbose /Users/ezio/Library/Arduino15/packages/digistump/hardware/avr/1.6.8/libraries/Digispark_Examples/Start/Start.ino
Using board 'digispark-tiny' from platform in folder: /Users/ezio/Library/Arduino15/packages/digistump/hardware/avr/1.6.8
Using core 'tiny' from platform in folder: /Users/ezio/Library/Arduino15/packages/digistump/hardware/avr/1.6.8
Detecting libraries used...
/Users/ezio/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -flto -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=16500000L -DARDUINO=10812 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR -I/Users/ezio/Library/Arduino15/packages/digistump/hardware/avr/1.6.8/cores/tiny -I/Users/ezio/Library/Arduino15/packages/digistump/hardware/avr/1.6.8/variants/digispark /var/folders/c4/z92mlq012sb_p89mpfdx5k900000gn/T/arduino_build_771399/sketch/Start.ino.cpp -o /dev/null
Generating function prototypes...
/Users/ezio/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -flto -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=16500000L -DARDUINO=10812 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR -I/Users/ezio/Library/Arduino15/packages/digistump/hardware/avr/1.6.8/cores/tiny -I/Users/ezio/Library/Arduino15/packages/digistump/hardware/avr/1.6.8/variants/digispark /var/folders/c4/z92mlq012sb_p89mpfdx5k900000gn/T/arduino_build_771399/sketch/Start.ino.cpp -o /var/folders/c4/z92mlq012sb_p89mpfdx5k900000gn/T/arduino_build_771399/preproc/ctags_target_for_gcc_minus_e.cpp
/Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /var/folders/c4/z92mlq012sb_p89mpfdx5k900000gn/T/arduino_build_771399/preproc/ctags_target_for_gcc_minus_e.cpp
Sto compilando lo sketch...
/Users/ezio/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -flto -MMD -mmcu=attiny85 -DF_CPU=16500000L -DARDUINO=10812 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR -I/Users/ezio/Library/Arduino15/packages/digistump/hardware/avr/1.6.8/cores/tiny -I/Users/ezio/Library/Arduino15/packages/digistump/hardware/avr/1.6.8/variants/digispark /var/folders/c4/z92mlq012sb_p89mpfdx5k900000gn/T/arduino_build_771399/sketch/Start.ino.cpp -o /var/folders/c4/z92mlq012sb_p89mpfdx5k900000gn/T/arduino_build_771399/sketch/Start.ino.cpp.o
Compiling libraries...
Compiling core...
[... cut ...]
> Suggested sleep time between sending pages: 7ms
> Whole page count: 103  page size: 64
> Erase function sleep duration: 721ms
parsing: 50% complete
> Erasing the memory ...
erasing: 55% complete
erasing: 60% complete
erasing: 65% complete
> Starting to upload ...
writing: 70% complete
writing: 75% complete
writing: 80% complete
> Starting the user app ...
running: 100% complete
>> Micronucleus done. Thank you!
```

Thank you for your promptly reply and fix!

And thanks to keeping this progect alive too,
Ezio

ArminJo, 2021-02-20:
@eziologico Please close the issue. Thanks!

eziologico, 2021-02-21:
Hi @ArminJo I don't know how to do it as I don’t have the close button, maybe @siliconchris1973 should do it that opened the issue.

ArminJo, 2021-02-21:
@siliconchris1973 Please close the issue. Thanks!