digistump/DigistumpArduino #111: Flash write error -1 has occured ...

jepunband, 2020-03-29:
hello, im getting this error with digispark using arduino during upload.

Sketch uses 2810 bytes (46%) of program storage space. Maximum is 6012 bytes.
Global variables use 106 bytes of dynamic memory.
Running Digispark Uploader...
Plug in device now... (will timeout in 60 seconds)
> Please plug in the device ... 
> Press CTRL+C to terminate the program.
> Device is found!
connecting: 16% complete
connecting: 22% complete
connecting: 28% complete
connecting: 33% complete
> Device has firmware version 2.2
> Device signature: 0x1e930b 
> Available space for user applications: 6522 bytes
> Suggested sleep time between sending pages: 7ms
> Whole page count: 102  page size: 64
> Erase function sleep duration: 714ms
parsing: 50% complete
> Erasing the memory ...
erasing: 55% complete
erasing: 60% complete
erasing: 65% complete
> Starting to upload ...
>> Flash write error -1 has occured ...
>> Please unplug the device and restart the program.

ArminJo, 2020-05-02:
What happened after restarting the program?

ArminJo, 2020-06-02:
Please close the Issue.

Kevitto, 2021-11-22:
Same issue here.  Scoured the internet and there are no solutions.  I'm trying to upload the blink example directly from the digistump Library, tried on two different usb hubs and three different attiny85s directly from Digistump.  

Any solutions?

MiranDaniel, 2021-12-29:
I am also having this issue

ArminJo, 2021-12-30:
Try [this ](https://github.com/ArminJo/DigistumpArduino) or [this](https://github.com/SpenceKonde/ATTinyCore) and check your fuses.

mander15, 2022-12-19:
Try taking out any spaces from your file name.  For some reason this happens for me when i have a space in my file name.