digistump/DigistumpArduino #112: Attiny-85 Uploading sketch without bootloader

HarryWickham, 2020-04-03:
I have written a sketch but when I try to compile it I get the error:

**Sketch uses 7520 bytes (125%) of program storage space. Maximum is 6012 bytes.
Global variables use 72 bytes of dynamic memory.
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
Error compiling for board Digispark (Default - 16.5mhz).**

And as the Attiny 85 has 8Kb of program storage, is there a way to upload this sketch without the 2Kb bootloader? Many thanks.

pfeerick, 2020-04-03:
I don't think so. And if you did, you would not be able to write another
program to it, as the bootload is what presents the Digispark as a USB
device and writes the new code to the Digispark. If you need to persue this
path, I would be looking at ISCP/ISP programming of the attiny85 on the
Digispark and possibly also using Spence Konde's ATTinyCore board support
package.

HarryWickham, 2020-04-03:
Many thanks for the reply, I will look into it in due course.

ArminJo, 2020-05-02:
Use the [optimized digistump version](https://github.com/ArminJo/DigistumpArduino), it enables around 20% more FLASH.

ArminJo, 2020-06-04:
Please could you be so kind and close the issue, since this repo is abandoned.
Thank you!
Armin

closed 2020-06-04 by HarryWickham