digistump/DigistumpArduino #114: PlatformIO framework conflicts toolchain (warning: "BIN" redefined in Print.h)

gnbl, 2020-04-25:
Unfortunately this project seems to be unmaintained?

This project/repository seems to be used in PlatformIO (PIO) as `framework-arduino-avr-digistump`. It redefines `BIN` ([Print.h#L37](https://github.com/digistump/DigistumpArduino/blob/master/digistump-avr/cores/tiny/Print.h#L37)) that was previously defined in PIO's `toolchain-atmelavr` (...\toolchain-atmelavr\avr\include\avr\iotnx5.h:55:0).
These defines are used in a different context. One solution might be a prefix to use a different "namespace".
There does not appear to be a related PR (https://github.com/digistump/DigistumpArduino/pulls).

The full warning is as follows:

```
In file included from C:\Users\user\.platformio\packages\framework-arduino-avr-digistump\cores\dtiny/Stream.h:24:0,
                 from C:\Users\user\.platformio\packages\framework-arduino-avr-digistump\cores\dtiny/TinyDebugSerial.h:31,
                 from C:\Users\user\.platformio\packages\framework-arduino-avr-digistump\cores\dtiny/WProgram.h:18,
                 from C:\Users\user\.platformio\packages\framework-arduino-avr-digistump\cores\dtiny/Arduino.h:4,
                 from src\main.cpp:20:
C:\Users\user\.platformio\packages\framework-arduino-avr-digistump\cores\dtiny/Print.h:37:0: warning: "BIN" redefined
 #define BIN 2
 ^
In file included from c:\users\user\.platformio\packages\toolchain-atmelavr\avr\include\avr\iotn85.h:38:0,
                 from c:\users\user\.platformio\packages\toolchain-atmelavr\avr\include\avr\io.h:428,
                 from c:\users\user\.platformio\packages\toolchain-atmelavr\avr\include\avr\interrupt.h:38,
                 from C:\Users\user\.platformio\packages\framework-arduino-avr-digistump\cores\dtiny/WProgram.h:8,
                 from C:\Users\user\.platformio\packages\framework-arduino-avr-digistump\cores\dtiny/Arduino.h:4,
                 from src\main.cpp:20:
c:\users\user\.platformio\packages\toolchain-atmelavr\avr\include\avr\iotnx5.h:55:0: note: this is the location of the previous definition
 #define BIN     7
 ^
```

I don't know how frameworks and toolchains are built for PlatformIO - this appears to be "upstream", but please let me know if there's a better place to report.

References:

- https://docs.platformio.org/en/latest/frameworks/arduino.html#platforms
- https://docs.platformio.org/en/latest/platforms/atmelavr.html#packages
- https://docs.platformio.org/en/latest/boards/atmelavr/digispark-tiny.html#board-atmelavr-digispark-tiny
- https://www.arduino.cc/reference/en/language/functions/communication/serial/print  
  >Serial.print(78, BIN) gives "1001110"

ArminJo, 2020-05-04:
This can not be resolved. Blame Atmel for defining BIN as as bit position (7) for the x5 ADC "Bipolar Input Mode".
I changed the message and it can be suppressed by commenting out [#define SUPPRESS_BIN_WARNING in line 39 in Print.h](https://github.com/ArminJo/DigistumpArduino/blob/2ceda03b4182a372486dad88500f80684ef79383/digistump-avr/cores/tiny/Print.h#L39)
The "Fix" is included in the new [version 1.6.8](https://github.com/ArminJo/DigistumpArduino) of Digistump avr core.

ArminJo, 2020-06-04:
Please close the issue, since this repo is abandoned.
Thank you!
Armin