digistump/DigistumpArduino #116: Question: how to compile DigiKeyboard.h without Arduino?

enty8080, 2020-05-09:
I can compile code for attiny digispark via avr-gcc / avr-g++, but when I include DigiKeyboard.h, avr-gcc or avr-g++ returns me a lot of errors. And also it uses Arduino.h((

ArminJo, 2020-05-09:
Do you expect code written for Arduino to run without it?

enty8080, 2020-05-09:
no, no... I mean:
```cpp
#include "DigiKeyboard.h"

int main(void) {
    DigiKeyboard.println("hello, world!");
}
```

enty8080, 2020-05-09:
I don't know how to make DigiKeyboard work not only with IDE

enty8080, 2020-05-09:
Is there any way to do this?

ArminJo, 2020-05-09:
You can not use Arduino libraries etc. without Arduino Core!!!
You must compile all the Arduino files from the core like e.g. wiring.c, Print.c and use **all** include files!
Or you create your own core or substitute the functions, the DigiKeyboard requires.

enty8080, 2020-05-09:
Do you know any keyboard/mouse HID libraries that works not only with Arduino IDE?

ArminJo, 2020-05-09:
No

enty8080, 2020-05-09:
I can’t believe that there is no way to program DigiSpark without Arduino :(

ArminJo, 2020-05-09:
Use the Atmel tools or Gcc with your own startup code. 
Or google for ATtiny85 programs maybe even in Github.
BTW, what is wrong with Arduino, it is mainly the startup code and the initialization of the millis() timer.

enty8080, 2020-05-09:
Ok. And the last question: how can I execute Arduino ide without GUI from terminal with specified programmer (Micronucleus)?

enty8080, 2020-05-09:
I mean something like this:
`./arduino -input sketch.ino -bootloader micronucleus`

ArminJo, 2020-05-09:
Arduino-cli

enty8080, 2020-05-09:
Do you know how to do something like this:
`./arduino -input sketch.ino -bootloader micronucleus` from Arduino CLI?

pfeerick, 2020-05-10:
What does the `-bootloader` parameter do? Given your earlier question, do you mean `programmer`? i.e. is that the equivalent GUI function? 

For the `arduino-cli`, the nearest to what I think you're after is [upload command](https://arduino.github.io/arduino-cli/commands/arduino-cli_upload/)... especially since which the digistump board support package you don't need to specify `micronucleus` as the programmer, since it is the platform default. You just use the normal `upload` command. 

Perhaps something like:

`arduino-cli upload -b digistump:avr:digispark-tiny /path/to/Sketch`

enty8080, 2020-05-10:
How to install digistump to arduino-cli?

enty8080, 2020-05-10:
But what should I do if I want to define my own programmer in arduino-CLI?

pfeerick, 2020-05-10:
https://arduino.github.io/arduino-cli/commands/arduino-cli_core_install/

You currently can't use a programmer... [according to the documentation](https://arduino.github.io/arduino-cli/platform-specification/#upload-using-an-external-programmer).

enty8080, 2020-05-10:
OK. thank you

enty8080, 2020-05-10:
Can I use DigiMouse with DigiKeyboard?

closed 2020-05-13 by enty8080