digistump/DigistumpArduino #120: error while uploading

jayrawatrj, 2020-06-01:
java.io.IOException: Cannot run program "___REMOVE___/launcher": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at java.lang.Runtime.exec(Runtime.java:485)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.runCommand(SerialUploader.java:383)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:197)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$UploadHandler.run(Editor.java:2047)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:247)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 11 more
An error occurred while uploading the sketch

ArminJo, 2020-06-02:
Thank you for sharing such precious information

jayrawatrj, 2020-06-03:
i am getting this error while using it on ubuntu 18.04

pfeerick, 2020-06-03:
Check the path referred to at the start of the error: `java.io.IOException: Cannot run program "REMOVE/launcher"`. Does the file `launcher` exist? If not, that is your problem. If it does, it would seem it's the wrong architecture for your device. Both 32 and 64 bit versions of x86 linux are supported... so if you are on a arm/aarch64 platform you are out of luck. 

If the file does exist, what is the output from `uname -a`, `file PATH-TO-LAUNCHER/launcher` and `ldd PATH-TO-LAUNCHER/launcher` ?