digistump/DigistumpArduino #123: Keyboard compile error

Mikolaj98p, 2020-06-02:
Hello,
When I compiling "hello wolrd" program for keyboard, compilation was terminated on platform io.
Can you help?

Thx.
https://gist.github.com/Mikolaj98p/9d7c29cba54c37491c81d78cb2dcb1ec

ArminJo, 2020-06-02:
Please try it with the new [version 1.6.8](https://github.com/ArminJo/DigistumpArduino) of Digistump avr core.
I included a change which could possibly fix your error.

pfeerick, 2020-06-03:
@ArminJo Since the OP is having problems in PlatformIO, some consideration as to how to use the your updated framework with PlatformIO is needed. It is possible to override the digistump package using `platform_packages` in the `platformio.ini`, but the layout of the current board support package used there expects the core volder for the tiny to be called `dtiny` (don't ask me _why_), and the cores, variants, etc, to be in the root. (i.e. [like this](https://github.com/pfeerick/DigistumpArduinoAvr), which I can use in PlatformIO with just in inclusion of `platform_packages = framework-arduino-avr-digistump @ https://github.com/pfeerick/DigistumpArduinoAvr`),  If you were able to make a zip file release available in that format, it would be a trivial change in the platformio configuration to use as it could be pointed at the zip file. Otherwise, if you think this is stable enough to completely replace the Digistump core now, it might be worth looking at getting it properly integrated into PlatformIO with a equest over at https://github.com/platformio/platform-atmelavr

ArminJo, 2020-06-03:
Hi Peter,
**thanks a lot of explaining the PlatformIO specials.**
[Here](https://github.com/platformio/platform-atmelavr/blob/9970ad03bfada6a8b1c18a264f99fcea0b98c9bf/builder/frameworks/arduino.py#L35) is the reason for the name dtiny.

Having your repo as reference, I created a [zip file](https://github.com/ArminJo/DigistumpArduino/blob/master/framework-arduino-avr-digistump-1.6.8.tar.gz) as you proposed. I named it .tar.gz since I found this is the name of the original file, but it is only a zip file. I do not know what the correct format is, which PlatformIO expects.

Maybe you can test it? This would be very helpful!

And in the next step I can document it and asking the PlatformIO guys to integrate it.

pfeerick, 2020-06-04:
lol... yeah, I expected it would be something to do with one of the other cores... but couldn't be bothered looking it up. Thanks for that! :)

Wow... I didn't expect that quick a response! Fantastic! And I will most certainly give it a try, and let you know if I have any issues. Might be a couple of days before I can really do much on that. Just as a quick test, I did `platform_packages = framework-arduino-avr-digistump @ https://github.com/ArminJo/DigistumpArduino/raw/master/framework-arduino-avr-digistump-1.6.8.zip` in a `platformio.ini` and tried re-compiling a simple blink sketch with no issues at all.

btw, am I right thinking the reason there's no `LED_BUILTIN` for the ... plain ... digispark because of the modelA/modelB or rev1 vs rev2/4 differences in LED connection?

ArminJo, 2020-06-04:
Thanks for the trigger, I included the LED_BUILTIN definition and changed Start to Blink, which helps to decide if you have a Model B. Have you ever seen such Model B? I have around 20 boards but all Model A.

pfeerick, 2020-06-12:
Thanks for that! No I haven't. Mine are the `Kickstarter Special Editions` and even they are rev2 boards, so I don't known about the rev1 board... maybe they went to beta testers? I suspect there will be no complaints with PB1 ;)

ArminJo, 2020-06-12:
Thanks for the feedback 👍

Mikolaj98p, 2020-06-14:
Ok now it's work

gigawatts, 2020-11-30:
What is the current status of this? Is there a documented work-around?

I am getting similar compile errors (as well as a bunch of other warnings) in PlatformIO, such as:
`C:\Users\me\.platformio\packages\framework-arduino-avr-digistump\libraries\DigisparkKeyboard/DigiKeyboard.h:59:1: m 'int' to 'char' inside { } [-Wnarrowing]`

ArminJo, 2020-11-30:
Try https://github.com/ArminJo/DigistumpArduino

gigawatts, 2020-12-01:
@ArminJo Thank you, that works great :)

I guess this (official) repo has been abandoned then?

ArminJo, 2021-02-20:
@Mikolaj98p Please close the issue. Thanks!