digistump/DigistumpArduino #126: Old version of micronucleus pakackgeed with arudino

Kilill, 2020-07-06:
The micronucleus version referenced in the "package_digistump_index.json" file is 2.0a4 the latest version from micronucleus is 2.04
Time for an upgrade ?

ArminJo, 2020-07-06:
Do you know, whre to find the binaries of this version?

Kilill, 2020-07-06:
No had to compile them myself from the micronuclues code

ArminJo, 2020-07-06:
Do you have it on Dropbox or GoogleDrive or...?

Kilill, 2020-07-06:
No just cloned it from the repo:https://github.com/micronucleus/micronucleus

reppad, 2020-11-20:
With an (almost) up-to-date firmware, this cause the failure of sketch upload and display the following message when trying to upload from Arduino IDE or Platform IO on Ubuntu 20.04 :

_Warning: device with unknown new version of Micronucleus detected.
This tool doesn't know how to upload to this new device. Updates may be available.
Device reports version as: 2.2_

quinot, 2021-01-17:
Also, interestingly, it looks like the micronucleus tool included in the Digistrump core purports (from filenames) to be 2.0a4, but is actually an older release, and does not support newer Micronucleus firmware (version 2).

ArminJo, 2021-02-04:
@Kilill I just packaged a [new version](https://github.com/ArminJo/DigistumpArduino/) with the new micronucleus binaries for all platforms - not requiring launcher any more, see [platform.txt](https://github.com/ArminJo/DigistumpArduino/blob/092538c0654cc561c8492cd21c7c50fde5f78c1e/digistump-avr/platform.txt#L96) 😀- and new bootloaders.

Can you close the Issue then?
Thanks

Kilill, 2021-02-05:
Sure :-) will not be able to test if for bit though

closed 2021-02-05 by Kilill