digistump/DigistumpArduino #127: At Compile time

Palingenesis, 2020-08-06:
I have had to move some files not currently using ( *.h and *.cpp) to a sub folder.
I think,
If a library/header file is in same folder as *.ino file.
Compiler reads it even if it has not been #included  in the *.ino file.

ArminJo, 2020-08-06:
which files did you moved?

pfeerick, 2020-08-07:
This sounds like an Arduino IDE issue, not a issue with the DigistumpArduino BSP. AFAIK, during compilation, the Arduino IDE preprocessor combines all `.cpp` and `.h` files in the sketchbook folder together with the main `.ino` file to effectively make one file. In other words, if it's visible as a tab in the Arduino IDE, it'll be included in the compile, regardless of if you specifically include it or not.

Palingenesis, 2020-08-07:
Probably tunnel vision on my part, I am no pro. just a retired hobbyist.
The issue is the header in the directory I haven't included in my code, it calls <Wire.h>

I take it you should not have files in the same folder as the *.ino file if you are not using them.
Even if you don't include them?

I will close this comment.
Tim

closed 2020-08-07 by Palingenesis

ArminJo, 2020-08-07:
*.h files does not matter. If they are not included, they are not compiled.

Palingenesis, 2020-08-07:
As I thought **ArminJo**, it must be as **pfeerick** says about the IDE.
I have posted the question on the Arduino forum.

My current project I test the code on a NANO and the ATTiny85.
So I have:
`#ifdef DEVICE_NANO
#include <Wire.h>
#else	//	DEVICE_ATTINY85
#include <USIWire.h>	//	https://github.com/puuu/USIWire
#endif
`
If I have the library for the DS1307 clock (I want to shorten it for the ATTiny85) in the same folder as the *.ino file, it wont compile for the ATTiny85, because the DS1307 Library includes the Wire library.
Even if I haven't included the DS1307 library in my *.ino file.

I learned something new today. (Use a sub-folder for custom libraries with Arduino IDE)

Thank you every one for putting me in the right direction.
Tim