digistump/DigistumpArduino #128: Unable to Find SoftwareSerial.h Include

djayor, 2020-08-17:
Environment: macOS 10.15.6; Arduino IDE 1.8.12; Digistump AVR Boards v1.6.7
Board: Digispark (Default - 16.5mhz) - Programmer: micronucleus

Hi, every time that I try to compile including a library for arduino, like SoftwareSerial.h I get a error:

```
Detecting libraries used...
/Users/melo/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=16500000L -DARDUINO=10812 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR -I/Users/melo/Library/Arduino15/packages/digistump/hardware/avr/1.6.7/cores/tiny -I/Users/melo/Library/Arduino15/packages/digistump/hardware/avr/1.6.7/variants/digispark /var/folders/tm/k68l_3wn5p18wh624ky_rzpc0000gn/T/arduino_build_428704/sketch/aTTiny85.ino.cpp -o /dev/null
Alternatives for SoftwareSerial.h: []
ResolveLibrary(SoftwareSerial.h)
  -> candidates: []
aTTiny85:3:10: fatal error: SoftwareSerial.h: No such file or directory
 #include <SoftwareSerial.h>
```

This is a example of the code:

```
#include <SoftwareSerial.h>

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

I already try to copy the *.h to the root folder of my *.ino, and change:
```
#include "SoftwareSerial.h"
```
But the compiler cannot find the file anyway.

ArminJo, 2020-08-17:
Try https://github.com/ArminJo/DigistumpArduino.

djayor, 2020-08-17:
Well it work anyway without the lines that I remove.
But now I have another problem, when I add "DigiKeyboard.h" library in the same program that have:

#include <SoftSerial.h>
OR
#include <TinyPinChange.h>

I get errors. If I compile this individual libraries isolated (different programs and folders) I can compile.

```
Error while detecting libraries included by /Users/melo/Library/Arduino15/packages/digistump/hardware/avr/1.6.7/libraries/DigisparkKeyboard/usbdrvasm.S
/var/folders/tm/k68l_3wn5p18wh624ky_rzpc0000gn/T/arduino_build_906347/libraries/DigisparkKeyboard/usbdrvasm.S.o: In function `__vector_2':
/Users/melo/Library/Arduino15/packages/digistump/hardware/avr/1.6.7/libraries/DigisparkKeyboard/usbdrvasm165.inc:41: multiple definition of `__vector_2'
/var/folders/tm/k68l_3wn5p18wh624ky_rzpc0000gn/T/arduino_build_906347/libraries/DigisparkTinyPinChange/TinyPinChange.cpp.o:/Users/melo/Library/Arduino15/packages/digistump/hardware/avr/1.6.7/libraries/DigisparkTinyPinChange/TinyPinChange.cpp:57: first defined here
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Digispark (Default - 16.5mhz).
```

ArminJo, 2020-08-17:
You know what `multiple definition of `__vector_2'` means?

djayor, 2020-08-17:
I don't know, but if I just add these 3 libraries to the program, I get that error.
I believe that one of the variables mentioned in this libraries is repeated.

ArminJo, 2020-08-17:
In short vector2 is for `Pin Change Interrupt Request 0`. 
So multiple libraries try to use the same hardware resource the vector2.
This is a common problem with libraries using limited hardware resources and cannot be solved.

djayor, 2020-08-18:
Oh thanks, this makes sense.
I solve the problem using this library, instead of "SoftSerial.h":
https://github.com/J-Rios/Digispark_SoftSerial-INT0
And instead of "DigiKeyboard.h" I used:
https://github.com/adafruit/Adafruit-Trinket-USB/tree/master/TrinketHidCombo

Works fine this way.

closed 2020-08-18 by djayor

Harshit-Mashru, 2025-04-17:
THANK YOU DAVID!! You saved my project! @djayor