digistump/DigistumpArduino #129: I don't recognize the serial port in the arduino interface. I have a version of windows 10 and arduino 1.8.8

alemori, 2020-09-25:
Install the drive shown on the site and when I insert the board in the usb port to record the sketch, although it records after not running the blink and I leave that windows did not recognize the device

ArminJo, 2020-09-25:
Try [this](https://github.com/ArminJo/DigistumpArduino)

alemori, 2020-09-26:
I went to the page in question, and updated board and drive but windows still gives me the error that does not recognize the driver.

ArminJo, 2020-09-26:
This sound like a windows problem. Please consider to reinstall your windows.

pfeerick, 2020-09-27:
Are you sure it's a driver issue? You don't program the digispark by setting a serial port, because it doesn't appear as one. It only appears as a USB device for five seconds after plugging it in, hence why you are _prompted_ to plug it in _during_ the upload process. 

What is the exact message (and when does it appear) that makes you think ...
> windows still gives me the error that does not recognize the driver.

... as perhaps your interpretation of what is happening is not what is actually happening?

alemori, 2020-09-30:
When I connect the Digispark I get a recognition error of the device, that's why I thought it was the drive. I bought another one and was able to move on, probably the board was damaged at some point during the tests, now I connect it through an extension of USB ports. Thank you very much.

ArminJo, 2021-02-20:
@alemori Please close the issue. Thanks!