digistump/DigistumpArduino #130: Flash write error -1

AlienWolfX, 2020-11-02:
Upon uploading the code to the Digispark i get the Flash write error -1 
Here is my log:
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Allen Cruiz\AppData\Local\Arduino15\packages -hardware C:\Users\Allen Cruiz\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Allen Cruiz\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Allen Cruiz\Documents\Arduino\libraries -fqbn=digistump:avr:digispark-pro -ide-version=10813 -build-path C:\Users\ALLENC~1\AppData\Local\Temp\arduino_build_296592 -warnings=none -build-cache C:\Users\ALLENC~1\AppData\Local\Temp\arduino_cache_511337 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\Allen Cruiz\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5 -prefs=runtime.tools.avr-gcc-4.8.1-arduino5.path=C:\Users\Allen Cruiz\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5 -prefs=runtime.tools.micronucleus.path=C:\Users\Allen Cruiz\AppData\Local\Arduino15\packages\digistump\tools\micronucleus\2.0a4 -prefs=runtime.tools.micronucleus-2.0a4.path=C:\Users\Allen Cruiz\AppData\Local\Arduino15\packages\digistump\tools\micronucleus\2.0a4 -verbose C:\Users\Allen Cruiz\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\libraries\DigisparkKeyboard\examples\Keyboard\Keyboard.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Allen Cruiz\AppData\Local\Arduino15\packages -hardware C:\Users\Allen Cruiz\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Allen Cruiz\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Allen Cruiz\Documents\Arduino\libraries -fqbn=digistump:avr:digispark-pro -ide-version=10813 -build-path C:\Users\ALLENC~1\AppData\Local\Temp\arduino_build_296592 -warnings=none -build-cache C:\Users\ALLENC~1\AppData\Local\Temp\arduino_cache_511337 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\Allen Cruiz\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5 -prefs=runtime.tools.avr-gcc-4.8.1-arduino5.path=C:\Users\Allen Cruiz\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5 -prefs=runtime.tools.micronucleus.path=C:\Users\Allen Cruiz\AppData\Local\Arduino15\packages\digistump\tools\micronucleus\2.0a4 -prefs=runtime.tools.micronucleus-2.0a4.path=C:\Users\Allen Cruiz\AppData\Local\Arduino15\packages\digistump\tools\micronucleus\2.0a4 -verbose C:\Users\Allen Cruiz\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\libraries\DigisparkKeyboard\examples\Keyboard\Keyboard.ino
Using board 'digispark-pro' from platform in folder: C:\Users\Allen Cruiz\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7
Using core 'pro' from platform in folder: C:\Users\Allen Cruiz\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7
Detecting libraries used...
"C:\\Users\\Allen Cruiz\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -w -x c++ -E -CC -mmcu=attiny167 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_DIGISPARKPRO -DARDUINO_ARCH_AVR "-IC:\\Users\\Allen Cruiz\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\cores\\pro" "-IC:\\Users\\Allen Cruiz\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\variants\\pro" "C:\\Users\\ALLENC~1\\AppData\\Local\\Temp\\arduino_build_296592\\sketch\\Keyboard.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for DigiKeyboard.h: [DigisparkKeyboard]
ResolveLibrary(DigiKeyboard.h)
  -> candidates: [DigisparkKeyboard]
"C:\\Users\\Allen Cruiz\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -w -x c++ -E -CC -mmcu=attiny167 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_DIGISPARKPRO -DARDUINO_ARCH_AVR "-IC:\\Users\\Allen Cruiz\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\cores\\pro" "-IC:\\Users\\Allen Cruiz\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\variants\\pro" "-IC:\\Users\\Allen Cruiz\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\libraries\\DigisparkKeyboard" "C:\\Users\\ALLENC~1\\AppData\\Local\\Temp\\arduino_build_296592\\sketch\\Keyboard.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
"C:\\Users\\Allen Cruiz\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -w -x c++ -E -CC -mmcu=attiny167 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_DIGISPARKPRO -DARDUINO_ARCH_AVR "-IC:\\Users\\Allen Cruiz\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\cores\\pro" "-IC:\\Users\\Allen Cruiz\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\variants\\pro" "-IC:\\Users\\Allen Cruiz\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\libraries\\DigisparkKeyboard" "C:\\Users\\Allen Cruiz\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\libraries\\DigisparkKeyboard\\oddebug.c" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
"C:\\Users\\Allen Cruiz\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -w -x c++ -E -CC -mmcu=attiny167 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_DIGISPARKPRO -DARDUINO_ARCH_AVR "-IC:\\Users\\Allen Cruiz\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\cores\\pro" "-IC:\\Users\\Allen Cruiz\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\variants\\pro" "-IC:\\Users\\Allen Cruiz\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\libraries\\DigisparkKeyboard" "C:\\Users\\Allen Cruiz\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\libraries\\DigisparkKeyboard\\osccal.c" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
"C:\\Users\\Allen Cruiz\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -w -x c++ -E -CC -mmcu=attiny167 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_DIGISPARKPRO -DARDUINO_ARCH_AVR "-IC:\\Users\\Allen Cruiz\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\cores\\pro" "-IC:\\Users\\Allen Cruiz\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\variants\\pro" "-IC:\\Users\\Allen Cruiz\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\libraries\\DigisparkKeyboard" "C:\\Users\\Allen Cruiz\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\libraries\\DigisparkKeyboard\\usbdrv.c" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
"C:\\Users\\Allen Cruiz\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -w -x c++ -E -CC -mmcu=attiny167 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_DIGISPARKPRO -DARDUINO_ARCH_AVR "-IC:\\Users\\Allen Cruiz\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\cores\\pro" "-IC:\\Users\\Allen Cruiz\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\variants\\pro" "-IC:\\Users\\Allen Cruiz\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\libraries\\DigisparkKeyboard" "C:\\Users\\Allen Cruiz\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\libraries\\DigisparkKeyboard\\usbdrvasm.S" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Error while detecting libraries included by C:\Users\Allen Cruiz\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\libraries\DigisparkKeyboard\usbdrvasm.S
Generating function prototypes...
"C:\\Users\\Allen Cruiz\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -w -x c++ -E -CC -mmcu=attiny1 [... cut ...]
connecting: 33% complete
> Device has firmware version 2.2
> Device signature: 0x1e930b 
> Available space for user applications: 6522 bytes
> Suggested sleep time between sending pages: 7ms
> Whole page count: 102  page size: 64
> Erase function sleep duration: 714ms
parsing: 50% complete
> Erasing the memory ...
erasing: 55% complete
erasing: 60% complete
erasing: 65% complete
> Starting to upload ...
>> Flash write error -1 has occured ...
>> Please unplug the device and restart the program.

ArminJo, 2020-11-02:
Thanks for the detailed information.
Try https://github.com/ArminJo/DigistumpArduino

ArminJo, 2021-02-20:
@AlienWolfX Please close the issue. Thanks!

closed 2024-08-26 by AlienWolfX