digistump/DigistumpArduino #132: UPDATE avr-gcc

Denial-Steinberg, 2021-02-03:
PLEASE UPDATE avr-gcc !!!

pfeerick, 2021-02-04:
Why? From what version? To what version? What newer version is compatible? Have you tried the more update to date fork at https://github.com/ArminJo/DigistumpArduino ?

ArminJo, 2021-02-20:
@Denial-Steinberg Please close the issue. Thanks!

DeuxVis, 2022-02-24:
@neatc0der I would try with an old arduino version if I were you, released around the date of the last update to those "Digistump SAM Boards" package.

Or maybe try with platformio ? https://docs.platformio.org/en/latest/boards/atmelsam/digix.html

aaronw2, 2022-10-14:
Please update it. I get collect2: error: ld terminated with signal 11 [Segmentation fault], core dumped when using the supplied 4.8.1. If I manually link it using the latest 7.3.0 it links without any issue.

/home/xxxxxx/.arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-gcc -Os -Wl,--gc-sections -mmcu=attiny85 -o /tmp/arduino_build_698262/ServoSaver.ino.elf /tmp/arduino_build_698262/sketch/ServoSaver.ino.cpp.o /tmp/arduino_build_698262/libraries/DigisparkTinyPinChange/TinyPinChange.cpp.o /tmp/arduino_build_698262/libraries/DigisparkSoftSerial/SoftSerial.cpp.o /tmp/arduino_build_698262/libraries/DigisparkSoftRcPulseIn/SoftRcPulseIn.cpp.o /tmp/arduino_build_698262/core/pins_arduino.c.o /tmp/arduino_cache_505193/core/core_digistump_avr_digispark-tiny8_91b9000a5bfd673e7bfcbaf492b2faae.a -L/tmp/arduino_build_698262 -lm
collect2: error: ld terminated with signal 11 [Segmentation fault], core dumped