digistump/DigistumpArduino #133: error=13, Permission denied

MRezaNasirloo, 2021-02-07:
OS: macOS Catalina
Arduino: 1.8.13
digistump: 1.7.1

stack trace: 
```
Arduino: 1.8.13 (Mac OS X), Board: "Digispark, 16 MHz - No USB, Default (6650 bytes avail.)"

Sketch uses 308 bytes (4%) of program storage space. Maximum is 6650 bytes.
Global variables use 9 bytes (1%) of dynamic memory, leaving 503 bytes for local variables. Maximum is 512 bytes.
java.io.IOException: Cannot run program "/Library/Arduino15/packages/digistump/tools/micronucleus/2.5/micronucleus": error=13, Permission denied
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at java.lang.Runtime.exec(Runtime.java:485)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.runCommand(SerialUploader.java:383)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:197)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$UploadHandler.run(Editor.java:2055)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: error=13, Permission denied
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:247)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 11 more
An error occurred while uploading the sketch

```

Looks like the `Library/Arduino15/packages/digistump/tools/micronucleus/2.5/micronucleus` file does not have execution permission

Fixed it by:
`chmod +x Library/Arduino15/packages/digistump/tools/micronucleus/2.5/micronucleus`

ArminJo, 2021-02-08:
Thanks for reporting. 🥇 
I have fixed it now for all platforms.

ArminJo, 2021-02-20:
@MRezaNasirloo Can you please close the issue?

closed 2021-02-21 by MRezaNasirloo