digistump/DigistumpArduino #134: Run error -1

Skingo12, 2021-02-18:
Hi,

I was wondering if I could get some help. When I try to upload to the digispark I get a run error -1. 
Here is my log:
Sketch uses 3208 bytes (53%) of program storage space. Maximum is 6012 bytes.
Global variables use 99 bytes of dynamic memory.
Running Digispark Uploader...
Plug in device now... (will timeout in 60 seconds)
> Please plug in the device ... 
> Press CTRL+C to terminate the program.
> Device is found!
connecting: 16% complete
connecting: 22% complete
connecting: 28% complete
connecting: 33% complete
> Device has firmware version 1.6
> Available space for user applications: 6012 bytes
> Suggested sleep time between sending pages: 8ms
> Whole page count: 94  page size: 64
> Erase function sleep duration: 752ms
parsing: 50% complete
> Erasing the memory ...
erasing: 55% complete
erasing: 60% complete
erasing: 65% complete
> Starting to upload ...
writing: 70% complete
writing: 75% complete
writing: 80% complete
> Starting the user app ...
>> Run error -1 has occured ...
>> Please unplug the device and restart the program. 

Thanks in advance,
Skingo12

ArminJo, 2021-02-18:
Please browse the old issues for a solution.

Skingo12, 2021-02-19:
Thanks for the quick response, I tried to install your alternative board manager but got a java error.  Now I get a new error which I think is a arduino IDE problem. I'll take it up with them but I would greatly appreciate it if you could help with this:

Arduino: 1.8.13 (Mac OS X), Board: "Digispark (Default - 16.5mhz)"

Sketch uses 302 bytes (5%) of program storage space. Maximum is 6012 bytes.
Global variables use 9 bytes of dynamic memory.
java.io.IOException: Cannot run program "___REMOVE___/launcher": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:621)
	at java.lang.Runtime.exec(Runtime.java:486)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.runCommand(SerialUploader.java:383)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:197)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$UploadHandler.run(Editor.java:2055)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:247)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 11 more
An error occurred while uploading the sketch

Thanks for your help

ArminJo, 2021-02-19:
Please remove the old core and install it again. If you get the same error please post it. There is no **launcher** any more.

Exci, 2021-08-07:
> 
> 
> Please browse the old issues for a solution.

I have the same issue as the first post but I couldn't find something related in the closed issues. I also tried the newer versions from your repo with the same result (except for 1.7.4 that gives a different error). What is odd is that the device was working for a bit, and at some point just started throwing out these ambiguous errors. Doesn't matter which code I'm using from the examples, it's always the same output. Could you clarify what old issue you mean?

ArminJo, 2021-08-09:
Whats the problem with your program not running after uploading? Just reset and it runs!

cold-inferno, 2021-08-12:
Got the same issue. Macbook 16" with a no name usb-c to usb-a dongle. That didn't work.
 Putting a usb 2.0 hub in between, however, made the issue go away.

USB-C dongle -> USB 2.0 Hub -> DigiSpark

it seems to be related to usb 2.0 devices not functioning correctly when plugged into usb 3 ports. Or the ports not responding well to usb 2.0 devices.

AlexMatter1512, 2022-03-31:
Same issue here. MacBook Pro 15 2017. I've tried with: 

- apple usb-C to usb-A dongle 
in this case it doesn't even recognise the board
- apple usb-C to usb-A dongle + multi usb 2.0 dongle
- apple usb-C to usb-A dongle + usb 2.0 extension cable
- apple usb-C to usb-A dongle + usb 2.0 extension cable + multi usb 2.0 dongle
- third party (aukey) usb-c hub
- third party (aukey) usb-c hub + extension cable
- third party (aukey) usb-c hub + extension cable + multi usb 2.0 dongle

in all these cases the output is the same:

```
Running Digispark Uploader...
Plug in device now... (will timeout in 60 seconds)
> Please plug in the device ... 
> Press CTRL+C to terminate the program.
> Device is found!
connecting: 16% complete
connecting: 22% complete
connecting: 28% complete
connecting: 33% complete
> Device has firmware version 1.6
> Available space for user applications: 6012 bytes
> Suggested sleep time between sending pages: 8ms
> Whole page count: 94  page size: 64
> Erase function sleep duration: 752ms
parsing: 50% complete
> Erasing the memory ...
erasing: 55% complete
erasing: 60% complete
erasing: 65% complete
> Starting to upload ...
writing: 70% complete
writing: 75% complete
writing: 80% complete
> Starting the user app ...
>> Run error -1 has occured ...
>> Please unplug the device and restart the program.
```

but as someone already said, unplugging the board and plugging it back again doesn't work.
I hope someone finds a solution.

ArminJo, 2022-03-31:
The solution is, to use the new kernel and uploader from the [ATTiny core](https://github.com/SpenceKonde/ATTinyCore)

JohannesBoll, 2022-08-10:
Had the same problem, I followed then your solution to install the new kernel and uploader. But now I get the following error:
`
Arduino: 1.8.19 (Mac OS X), Board: "ATtiny85 (Micronucleus / DigiSpark), Enabled, CPU (CPU frequency), 16.5 MHz, tuned off USB (USB), B.O.D. Disabled (saves power), Enabled, Upgrade (via USB), Reset" 
> Please plug in the device (will time out in 60 seconds) ... 
> Device is found!
connecting: 16% complete
connecting: 22% complete
connecting: 28% complete
connecting: 33% complete
> Device has firmware version 1.6
> Available space for user applications: 6012 bytes
> Suggested sleep time between sending pages: 8ms
> Whole page count: 94  page size: 64
> Erase function sleep duration: 752ms
parsing: 50% complete
> Erasing the memory ...
erasing: 55% complete
erasing: 60% complete
erasing: 65% complete
> Starting to upload ...
writing: 70% complete
writing: 75% complete
writing: 80% complete
> Starting the user app ...
>> Run error: Device not configured has occured ...
>> Please unplug the device and restart the program. 
the selected serial port >> Please unplug the device and restart the program. 
 does not exist or your board is not connected
`
Has someone any hint to solve this problem?

bart12k, 2023-08-29:
Running Digispark Uploader...
Plug in device now... (will timeout in 60 seconds)
> Please plug in the device ... 
> Press CTRL+C to terminate the program.
> Device is found!
connecting: 16% complete
connecting: 22% complete
connecting: 28% complete
connecting: 33% complete
> Device has firmware version 1.6
> Available space for user applications: 6012 bytes
> Suggested sleep time between sending pages: 8ms
> Whole page count: 94  page size: 64
> Erase function sleep duration: 752ms
parsing: 50% complete
> Erasing the memory ...
erasing: 55% complete
erasing: 60% complete
erasing: 65% complete
> Starting to upload ...
writing: 70% complete
writing: 75% complete
writing: 80% complete
> Starting the user app ...
>> Run error -1 has occured ...
>> Please unplug the device and restart the program. 
Please help me.

pfeerick, 2023-08-29:
@bart12k Have tried this core instead? https://github.com/ArminJo/DigistumpArduino