digistump/DigistumpArduino #154: Outdated Micronucleus version

6r1d, 2022-07-06:
Hello.

Recently, I've got a couple of Digispark boards from a friend. The interesting things about these boards was that Micronucleus version was updated on them, so I was not able to program them from Arduino IDE yet. There was an error message about an unknown version of Micronucleus, too.

I've proceeded with updating the Micronucleus further and that raised the version and again, blocked Arduino IDE from writing.

I'm sorry for not logging the error message exactly, but it was showing `micronucleus` itself complaining about the unknown version.

That gave me an idea: what if I replace the executable of Micronucleus and experiment? I've checked the [releases](https://github.com/micronucleus/micronucleus/releases) and picked `v2.6`: [micronucleus-cli-master-882e7b4a-x86_64-Linux.zip](https://github.com/micronucleus/micronucleus/releases/download/v2.6/micronucleus-cli-master-882e7b4a-x86_64-Linux.zip).

So, maybe it's a good time for that small update and some testing.

LightningStalker, 2022-07-16:
I had to compile the latest version of Microneucleus to get the boards to program. Nothing has been updated since 2017!?

pfeerick, 2022-07-16:
Correct... this repo is basically abandoned. You should be looking at https://github.com/ArminJo/DigistumpArduino for more up-to-date support for the Digispark and clones.

6r1d, 2022-07-17:
> You should be looking at https://github.com/ArminJo/DigistumpArduino
Thanks!